LJMUAstroecology/flirpy #2: Tau Camera API documentation

Hrayo712, 2020-02-26:
Hello!

I am currently working with the Tau2 camera. However, I see that the API is pretty different compared to the Boson or the Lepton (no grab() function for example). Could you share with me the workflow (pipeline) which should be followed to capture a frame and display it ?

Thanks!

jveitchmichaelis, 2020-02-26:
Hi,

Do you have just a Tau? Or something connected to it? The Tau itself supports a very limited form of readout over serial (and I think not even radiometric), which flirpy can deal with, but it's a bit experimental and slow. Otherwise you need an interface board like the TeAx ThermalGrabber.

Thanks

Hrayo712, 2020-02-26:
Hi,

Thanks the reply. I currently just have the Tau2 connected directly through serial (USB) . I will try what you suggest. Is there any other way then (outside of flirpy) to interface with the Tau programmatically ?

Thanks

jveitchmichaelis, 2020-02-26:
Oh ok if you have the thermal grabber (one of [these](https://thermalcapture.com/thermalcapture-grabber-usb/)?) then things are a bit better. I haven't finished the driver - I've got the logic to sync with the video stream and capture images, but I've not been able to figure out how to trigger the grabber to start streaming.

If you use TeAx's software to open the camera and then close it again, you can try (in flirpy):

```
with Tau() as cam:
    image = cam.grab_teax()
```

For some reason just opening the serial port doesn't quite work. There's some other connection step I'm missing. But once you've opened it in their software, it seems like you can just scoop up data.

I suspect the serial connection is super slow because it's going through TeAx's logic which has to repeat everything.

But... if you have Thermal Grabber, you can just use their library? Otherwise the Tau interface is reasonably well documented, just search for Tau Software IDD.

Hrayo712, 2020-02-26:
Hi,

Thanks a lot for your input, you have definitely given me things to work with and saved me some time, as I am currently a little bit short in time for integrating this camera into my project. 

Just to be sure I understood. The tau2 core can be interfaced directly (serial) via the software IDD. However the thermal grabber provides an extra interface to deal with data easily, providing the TeAx library to deal with It programmatically. I am assuming that all the communication (configuration/ data) which can be done directly to the Tau2 (without the thermal grabber) can also be done via the grabber (this is what I assume you mean by doubling the logic), am I right?

Sorry for so many questions and thanks a lot!

jveitchmichaelis, 2020-02-26:
No worries.

The Tau2 itself only provides a low level electrical interface (unsurprisingly, this is found in the Tau2 Electrical IDD :) ) which is a high density connector that has a direct serial control interface, as well as parallel and LVDS data interfaces. Not particularly useful on its own, you need an interface board which connects to that to read out data. Typically this is either a microcontroller or an FPGA.

The ThermalGrabber is (I believe) an FPGA -- it might be an Arm chip -- with an FTDI USB controller. Their SDK lets you do everything, more or less. And yeah, I think when you send a command that doesn't relate to image capture, it just relays it to the camera core and returns the result. You can use any serial terminal you like to talk to the camera (for control purposes). The images are also returned over serial by the way.

I've not used the SDK directly, though I've looked through the code.

Hrayo712, 2020-02-27:
Thanks a lot for your support,

I have been through the Thermal Capture Grabber SDK. They provide a mean to interface with the thermal camera via a v4l2loopback kernel module, running a daemon to access it as a regular webcam /dev/video0. Then access can be done with really simple python + OpenCV code.

I figured you could be interested in it, also wanted to say thanks. 

Regards from the Netherlands!

closed 2020-02-27 by Hrayo712