LJMUAstroecology/flirpy #6: Flir Tau2 support

YashRunwal, 2020-04-29:
In the docs it is mentioned that this library supports flir Tau Cameras. Does this support Flir Tau2 Cameras too?

jveitchmichaelis, 2020-04-29:
Yes, the Tau 2 is supported. Though please note the limitations in the readme - unless you have a readout board, image capture will be very slow.

jveitchmichaelis, 2020-04-29:
I've updated the readme to reflect that _only_ the Tau 2 has been tested.

https://github.com/LJMUAstroecology/flirpy/commit/9a35fb1a4ceca557acea165543acf5cd9d900f1e

closed 2020-04-29 by jveitchmichaelis

YashRunwal, 2020-04-29:
Thank You so much... I am sorry but I want to know whether you know a ROS driver for Flir Tau 2 is available or not. I know this is a different issue, but I was unsure as to whether i should open a different issue or not.

jveitchmichaelis, 2020-04-29:
Not sure. I don't think there is. You'd probably have to write one, but it wouldn't be complicated if you're just publishing images. Do you have an adapter for the camera core?

YashRunwal, 2020-04-30:
I have a VPC (Video Power Communication) and Camera Link Expansion Board. Is this what you mean?

jveitchmichaelis, 2020-05-10:
Something that will let you capture images digitally (i.e. not analogue), so camera link would be fine. However, this is somewhat out of flirpy's scope at the moment. I might provide some ROS examples in the future when ROS Noetic goes live.