LJMUAstroecology/flirpy #7: RAW-16 Not Working

amirgholami, 2020-05-03:
Hi,

First let me thank you for creating this repository. It is very useful and the code is very clean. I am trying to get the RAW pre AGC 16-bit data, but using the provided instructions I am only getting uint8:

camera = Boson()
camera.setup_video(device_id=2)
image = camera.grab()
print(image.dtype)
uint8

I did check that the following lines return 0 so I am not sure why this is happening:

# The order of these calls matters!
self.cap.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter_fourcc(*"Y16 "))
self.cap.set(cv2.CAP_PROP_CONVERT_RGB, False)

P.S: I am running this on MAC

jveitchmichaelis, 2020-05-03:
>  I am running this on MAC

Unfortunately this is the problem - 16 bit video support is not great on OS X unfortunately. You might have some luck with a generic UVC driver, but you can't use OpenCV.

Take a look at https://github.com/groupgets/libuvc for example. It should support Y16. Maybe also you could try gstreamer? That may be a good route actually, I'll have a look into that (though it would require compiling everything from scratch).

amirgholami, 2020-05-04:
That is right, it seems the problem is OSX. I tested it on Linux and the 16-bit works correctly.

shinya-ohtani, 2022-02-21:
> You might have some luck with a generic UVC driver, 

On macOS before macOS 12 (macOS 11.6.2 or earlier to be exact), libuvc works. However, on macOS 12, libuvc does not work at all. When trying to open the camera device, uvc_open() results in "uvc err: Access denied (-3)".

more details in this Issue:
https://github.com/groupgets/purethermal1-uvc-capture/issues/56