LJMUAstroecology/flirpy #8: PureThermal 2 - FLIR Lepton Temperature Measurement

Varat7v2, 2020-05-05:
Hi @jveitchmichaelis, I am using PureThermal 2 - FLIR Lepton camera for thermal imaging. But when I capture the image, lot of meta-data are missing. How can I capture image or use my camera to make it compatible with exiftool? When I run exiftool <on-my-image> its just showing the following details:

> ExifTool Version Number : 10.80
File Name : my_photo-1.jpg
Directory : .
File Size : 5.7 kB
File Modification Date/Time : 2020:05:04 20:28:10+05:30
File Access Date/Time : 2020:05:04 20:28:47+05:30
File Inode Change Date/Time : 2020:05:04 20:28:10+05:30
File Permissions : rw-rw-r--
File Type : JPEG
File Type Extension : jpg
MIME Type : image/jpeg
JFIF Version : 1.02
Resolution Unit : inches
X Resolution : 120
Y Resolution : 120
Image Width : 160
Image Height : 120
Encoding Process : Baseline DCT, Huffman coding
Bits Per Sample : 8
Color Components : 3
Y Cb Cr Sub Sampling : YCbCr4:2:2 (2 1)
Image Size : 160x120
Megapixels : 0.019

Also, how can I use flirpy with my camera to measure the temperature in degree Celsius?

Thank you.

jveitchmichaelis, 2020-05-05:
Hi, have you tried the examples here?

https://github.com/LJMUAstroecology/flirpy/blob/master/README.md#grab-images

The image metadata you've posted is 8 bit. So you're not capturing a raw image. If you're having trouble with exiftool, you should try their support forums directly.

Varat7v2, 2020-05-05:
@jveitchmichaelis yeah I tried the example. Its giving a numpy array of the image but when I divided it by 100 I am getting values in the range 70-90 in every pixels. And regarding telemetry, how can I enable them? I even tried https://github.com/groupgets/purethermal1-uvc-capture/tree/master/python/uvc-radiometry.py, I was confused with the output array as it does not seems to be in pixel-wise temperature.

Thank you for sharing the references, I will go through them.

jveitchmichaelis, 2020-05-05:
Is it a 16 bit array? Also, what platform are you using?

For example 30000 counts is (300 - 373.15) = 26.85 C.

Not sure how you enable telemetry, the pure thermal cameras I've used have always had it enabled.

Varat7v2, 2020-05-05:
I am running on Python3 in terminal shell. In my case at room temperature, the flirpy is giving me counts in the range 8000-9000 only. When I divided by 100 and subtracted by 273.15, all the values are coming in negative. Am I doing anything wrong?

jveitchmichaelis, 2020-05-05:
Sorry, I meant what OS. Are you using a Mac?

Not sure, do you get sensible results if you use the GroupGets gui?

Varat7v2, 2020-05-05:
No I am using Ubuntu 18.04. I haven't tried with groupgets gui. However, I had run it on uvc-radiometry.py, results doesn't seems to be sensible. If needed I will try with that one also.

jveitchmichaelis, 2020-05-05:
Yeah worth giving that a go, if that still doesn't work, there's probably a way to flash the firmware on the PT2 to the default.

Varat7v2, 2020-05-07:
Hi @jveitchmichaelis I tried [GetThermal](https://github.com/groupgets/GetThermal), but the temperature bar is not visible, so I am confused if its showing correct temperature profile.

Flirpy seems to be giving correct values but I am not getting values in the range you have mentioned above (i.e. 30000 counts). I am getting values in the range (7000 - 9000) and when I tested on different temperature object, those values are changing like:

- ICE --> (=7190)
- Normal water --> (=7694)
- Hot water --> (9298)
- Normal human body --> (=8317)

Can you please suggest me how to use these 16bit values that I am getting in this range?

Thank you.

jveitchmichaelis, 2020-05-07:
You know I may have seen this before. Try re-seating the camera on the PT2 board, it's possible you've just got a loose connection. Alternatively speak to GroupGets and ask how you can check if TLinear mode is enabled.

I suggest following this guide from FLIR which has the necessary precautions about handling the detector, it's quite sensitive: https://lepton.flir.com/wp-content/uploads/2015/06/PureThermal-2-Basic-Lepton-Features.pdf

Slides 9-10

Varat7v2, 2020-05-07:
I am using Lepton 3.0. In the [datasheet](https://groupgets-files.s3.amazonaws.com/lepton/3.5/Lepton%20Engineering%20Datasheet%20Rev200.pdf), it is mentioned that it does not have thermal radiometry configuration.

jveitchmichaelis, 2020-05-07:
Oh ok, then you can't get temperature measurements with your camera. That's what radiometry is for. Sorry about that. 

You can calibrate the camera yourself in principle, but you would need a blackbody calibration target at least, and ideally also a climate chamber.

Varat7v2, 2020-05-07:
Okay I will do the same.

Thank you @jveitchmichaelis for your help.

jveitchmichaelis, 2020-05-07:
Having done this with other cameras, I'll be honest - the price of a Lepton 3.5 is less than the cost of the calibration equipment and the time you'll spend doing it. A good blackbody target is at least $500.

closed 2020-05-07 by jveitchmichaelis