LJMUAstroecology/flirpy #9: lepton.find_video_device() does not find Lepton when picamera connected

zoltanszalontay, 2020-05-18:
https://github.com/LJMUAstroecology/flirpy/blob/9a35fb1a4ceca557acea165543acf5cd9d900f1e/flirpy/camera/lepton.py#L66

Automatic video capture device enumeration using lepton.find_video_device() on a Raspberry + Lepton 3.5 + PureThermal setup does not work for me when picamera is also connected.

I guess the problem is at the 66. 'i' is the number of order the video device has been found, which depends on the folder structure. However, cv2.videocapture(id) expects an id that is equivalent to the last figures of the filename of the device driver (as in /sys/class/video4linux/).

Background info:
Video devices:
['/sys/class/video4linux/video12', '/sys/class/video4linux/video1', '/sys/class/video4linux/video10', '/sys/class/video4linux/video2', '/sys/class/video4linux/video11', '/sys/class/video4linux/video0']

print d => "[1, 5]" based on filtering capabilities.

The correct id would be 0 (../video0) and cam = camera.setup_video(0) is working.

jveitchmichaelis, 2020-05-18:
Nice catch, yes the order of devices can't be guaranteed... Can probably replace it with something like:

`dev.append(int(device.split('video')[-1])) `

zoltanszalontay, 2020-05-18:
I was thinking about the same, except it is not guaranteed that the number of devices is less than 10 (one decimal figure).

jveitchmichaelis, 2020-05-18:
That should be OK, e.g.

```
> 'video100'.split('video')
['', '100']
```

Split returns the string, not the character.

zoltanszalontay, 2020-05-18:
I can confirm, that the patch above is working on my config. 

Thanks a lot!
Zoltan

jveitchmichaelis, 2020-05-18:
Thanks, glad it works - I'll patch this shortly!

jveitchmichaelis, 2020-05-18:
https://github.com/LJMUAstroecology/flirpy/commit/836800908b6a6e9e05682abf0b679f2c782a5eac

closed 2020-05-18 by jveitchmichaelis