LJMUAstroecology/flirpy #11: No output generated if output path contains special characters

LukasBommes, 2020-06-23:
Hey jveitchmichaelis,

I found another little bug in the split_seqs script (as in flirpy 0.1.1). I had the character "ä" (used quite often in German) in my output path and the script stopped generating any preview JPEGs or TIFFs, however without throwing any exception.

You can likely reproduce the error by running:
```
python split_seqs.py --input "*.SEQ" --output "outä"
```
It is not a big deal as I can just rename my paths. But just to let you know.

Regards,
Lukas

jveitchmichaelis, 2020-06-25:
Ah unicode fun... Are you running python 3?

LukasBommes, 2020-06-25:
Yes, I am running 3.6.7
Can you reproduce the error? Maybe, it is not even the fault of flirpy. Because, I get the same behavior with OpenCV's imwrite function (version 3.4.2) on my system. In case you can not reproduce the error you can probably just close this issue.

jveitchmichaelis, 2020-06-26:
So this works for me:

```python
import cv2
import numpy as np

test = np.array((100,100,3))
cv2.imwrite(test, "käfer.png")
```

I think I have a more recent version of OpenCV (e.g. `pip install python-opencv-headless`).