LJMUAstroecology/flirpy #13: No matching distribution for opencv-python-headless

basavarajamogi, 2020-07-10:
I am working on Lepton 3.5 camera using raspberry pi 4. I did clone the flirpy github repository.
I am getting following error.
Why it is? Any solution please..

pi@raspberrypi:~/flirpy $ pip install flirpy
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting flirpy
  Using cached flirpy-0.1.1-py2.py3-none-any.whl (10.2 MB)
Requirement already satisfied: tqdm<5.0,>=4.42 in /home/pi/.local/lib/python3.7/site-packages (from flirpy) (4.47.0)
Requirement already satisfied: pyserial<4.0,>=3.4 in /usr/lib/python3/dist-packages (from flirpy) (3.4)
ERROR: Could not find a version that satisfies the requirement opencv-python-headless<5.0,>=4.2 (from flirpy) (from versions: 3.4.2.16, 3.4.2.17, 3.4.3.18, 3.4.4.19, 3.4.6.27, 3.4.7.28, 4.0.1.24, 4.1.0.25, 4.1.1.26)
ERROR: No matching distribution found for opencv-python-headless<5.0,>=4.2 (from flirpy)

jveitchmichaelis, 2020-07-10:
I guess that version of OpenCV isn't available for arm yet. You can run `pip install flirpy --no-dependencies` and `pip install opencv-python-headless` separately.

basavarajamogi, 2020-07-10:
> 
> 
> I guess that version of OpenCV isn't available for arm yet. You can run `pip install flirpy --no-dependencies` and `pip install opencv-python-headless` separately.

Thank you, It did solve this problem...

closed 2020-07-10 by jveitchmichaelis