LJMUAstroecology/flirpy #14: Flirpy Lepton3.5 breakoutboard RPI4

basavarajamogi, 2020-07-10:
I am using RPI4 with breakoutboard and trying to grab images using Lepton 3.5 using flirpy.
I am getting following error when I am trying to run example program capture_lepton.py
Please let me know the solutions?
error:

Traceback (most recent call last):
  File "/home/pi/flirpy/examples/capture_lepton.py", line 4, in <module>
    image = cam.grab()
  File "/home/pi/.local/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 156, in grab
    self.setup_video(device_id)
  File "/home/pi/.local/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 97, in setup_video
    raise ValueError("Lepton not connected.")
ValueError: Lepton not connected.

jveitchmichaelis, 2020-07-11:
The SPI breakout board is currently unsupported, sorry. Or are you using the PureThermal?

basavarajamogi, 2020-07-11:
> The SPI breakout board is currently unsupported, sorry. Or are you using the PureThermal?

No, I am not using Pure thermal. I am using breakout board v 2.0. Is there any way out ?

jveitchmichaelis, 2020-07-11:
You'll have to use another library in the meantime. The place you bought the board from should be able to point in the right direction. Flir have some official documentation here:

https://lepton.flir.com/getting-started/raspberry-pi-lepton/

basavarajamogi, 2020-07-11:
> You'll have to use another library in the meantime. The place you bought the board from should be able to point in the right direction. Flir have some official documentation here:
> 
> https://lepton.flir.com/getting-started/raspberry-pi-lepton/

Yes, I tried them all but those are all c libraries. I am not anyway good in c, so just finding something in Python. On tithun I found Python codes for image capture but I was looking for something simple as flirpy. You have made flirpy for almost everything. I think only breakout board is left out.

jveitchmichaelis, 2020-07-11:
Have you tried the GroupGets library?
https://github.com/groupgets/pylepton

I should be able to add support for SPI in flirpy soon, but I don't use Leptons too often so it's not been a priority.

basavarajamogi, 2020-07-11:
> Have you tried the GroupGets library?
> https://github.com/groupgets/pylepton
> 
> I should be able to add support for SPI in flirpy soon, but I don't use Leptons too often so it's not been a priority unfortunately.

Yes, I tried that one... It works well....I am able to capture images but no temperature values, metadata for temperature

jveitchmichaelis, 2020-07-11:
Hmm OK, I'll see what I can do - I may have some time this weekend!

basavarajamogi, 2020-07-11:
> Hmm OK, I'll see what I can do - I may have some time this weekend!

Yes, That will be great for me sir... Thanks

jveitchmichaelis, 2020-07-11:
The GroupGets library should return radiometric values though, it says the images should be 12 bit - is that not what you get?

basavarajamogi, 2020-07-11:
I am just getting images...But not sure of it's 12 bit...

yasin-aydn, 2024-04-06:
> The SPI breakout board is currently unsupported

Hello,

Is breakout board still unsupported?