LJMUAstroecology/flirpy #17: How can I convert ats file to radiometric images?

monajalal, 2020-07-29:
I have video captures from FLIR SC8000 camera and I used ResearchIR to capture ats files. I now need to convert them to radiometric images/videos. How can I do so using your package?

jveitchmichaelis, 2020-07-29:
Hi, flirpy doesn't support ats files at the moment. It's mostly written for current camera cores. You could try the FLIR File SDK, but I've not tried it. 

http://flir.custhelp.com/app/answers/detail/a_id/3000/~/which-sdk-should-i-use%3F

Can you use ResearchIR to export your data? I would be surprised if it couldn't do the radiometric conversion for you.

monajalal, 2020-07-29:
@jveitchmichaelis thanks a lot for your prompt response. I am new to ResearchIR. I will go to lab and check that out. Could you please let me know what file format I should save the raw data to and what file format I should use for lossless video for accessing radiometric data?

Is your library independent of which FLIR thermal camera I will use? I am interested to know if I could use your code for SC8000 camera that I have.

jveitchmichaelis, 2020-07-29:
Lossless video is always going to be difficult because there are few codecs that support lossless 16-bit. This is essentially why FLIR has several proprietary formats like SEQ, ATS, etc. Your best bet is to not use video at all for radiometric data and just store the individual frames as 16-bit TIFFs - ResearchIR should be able to do that.  SEQ, for example, is just a stack of TIFF images with metadata. I guess ATS is similar.

https://flir.custhelp.com/app/answers/detail/a_id/2372/related/1

But you might need the higher tier version of ResearchIR to export?

> Is your library independent of which FLIR thermal camera I will use? I am interested to know if I could use your code for SC8000 camera that I have.

Unfortunately, no. All the cameras require different drivers for communication and image readout, so I can only support cameras that I have access to (e.g. the Tau, Lepton and Boson).

The SC8000 is now discontinued, but I think you can probably control it with the Atlas SDK: https://flir.custhelp.com/app/answers/detail/a_id/1275

That's at least free, but you'd have to write your own code for it.