LJMUAstroecology/flirpy #18: FFC control

nadavzil1, 2020-08-02:
Hi guys,
thank you for the great work!
Is there a way to control the Lepton 3.5 FFC with your library? I need to set the FFC mode to manual (so it won't happen every 3 minutes or 1.5 celsius but only when I call it). I would also like to change the number of integrated frames.

All the solutions I found online require programming in C which I don't know. Hopefully your project will finally solve my issue.

jveitchmichaelis, 2020-08-02:
Right now no, but this might be possible in Python using the UVC extension unit. But integrated frames you could always do in software, just do a median of however many you like?

EduardoRonchi, 2022-10-19:
Any chance to add the manual fcc operation for Lepton 3.5 sensors?

jveitchmichaelis, 2022-10-19:
Sorry, this still isn't supported and I don't think will be without some serious investment in a libusb wrapper (see above) which I don't have time for at the moment.