LJMUAstroecology/flirpy #27: Installing flirpy inside conda, split_seqs command is not recognized

GZuin, 2020-08-25:
Attempted to install flirpy inside of a conda environment on both Windows 10 and Ubuntu 18.04. In both cases, after installing through pip, no command split_seqs is found.

I can safely import flirpy inside of python, but after reading the split_seq example, I'm not sure what to do to split a seq file into images through python.

jveitchmichaelis, 2020-08-25:
This seems to be an issue with installing via pip. If you install using python setup.py install, it should place the script on your path (I just tested this in a new conda env on my Ubuntu machine). For now you can just copy the code in `scripts` and it'll work.

On Windows this is a pain because of the way that Windows interprets files. It will find it, but won't execute properly (this is on the todo list), so you should also just copy the code and run it from somewhere.

jveitchmichaelis, 2020-08-25:
So this is a poetry issue...

https://github.com/python-poetry/poetry/issues/241
https://github.com/python-poetry/poetry-core/pull/40

It seems like there is a lot of confusion over what a script is -  whether it is a "build" script or a user script that gets installed along with a package.

closed 2020-08-27 by GZuin

jveitchmichaelis, 2020-08-27:
@GZuin I need to push a new version shortly anyway, so I'll do it through Twine just as a precaution