LJMUAstroecology/flirpy #29: Cannot capture from lepton

elitearmedforce, 2020-09-25:
Hello,
I tried Flirpy today and noticed some weird issues. Honestly, I don't know how to fix it, please help!!!. (Details and screenshot attached below)

Device/installation details:
Host OS: Ubuntu 20.04.1 (dual boot)
FLIR device: FLIR Lepton 3.5
I/O hardware: Groupgets PureThermal 2
Opencv 4.3.0 installed on the host. Although not need if opencv-python-headless is installed
Python: 3.8.2 & PIP: v20.... something

Screenshot(tried both usb3.0 & usb 2.0)
[image]

jveitchmichaelis, 2020-09-25:
This is due to newer versions of OpenCV/Python. Can you try a `pip install --upgrade flirpy`?

elitearmedforce, 2020-09-25:
Appreciate ur immediate response. Just installed today, v0.2.2. FYI, using python3 and pip3. 

>This is due to newer versions of OpenCV/Python and the arguments to cv2.VideoCapture.set(), the True/False should be 1.0/0.0.

As per this comment, i understand that i should be using older version of flirpy?????

[image]

jveitchmichaelis, 2020-09-25:
Oh perhaps this was only fixed for the Boson driver not the Lepton. I'll take a look. For now just replace True with 1 and False with 0 where python is complaining.

closed 2020-09-25 by jveitchmichaelis

jveitchmichaelis, 2020-09-30:
FYI this is now pushed to PyPI v0.1.3