LJMUAstroecology/flirpy #30: Can

csedwar, 2020-09-30:


csedwar, 2020-09-30:
simply not live yet in the pip repo.  fixed by cloning the git repo and doing a pip install .

closed 2020-09-30 by csedwar

jveitchmichaelis, 2020-09-30:
Should be pushed now, sorry about that! v0.1.3

csedwar, 2020-09-30:
No worries at all! This toolkit is great.  It is already saving me a ton of time. Everything else out there is so painful!

Eventually I want to capture data from ~4-6 leptons and tag it with a GPS, so am awaiting the list enumeration as well! If I get a little time i'll take a stab at it. 

If you have suggestions for methods to do this efficiently and write out the data I'd be all ears. It appears there is some exif support here but its not super clear to me how to use it just yet.

jveitchmichaelis, 2020-09-30:
List enumeration in principle is easy, I've just not gotten around to it. The code now should be able to find all devices with a particular USB VID/PID, so it's just a matter of matching those to the devices returned by V4L (or DShow).

Writing data really depends on what sort of frame rates you're working with. Leptons are only 9Hz and the images are quite small, so you could just store them as single files. FLIR stores sequences as stacks (literally SEQ=sequence) of images with some header information. If you store your images as 16-bit (radiometric) TIFFs then you can add geotags in the EXIF, that's probably what I'd recommend. You can use whatever encoding you feel like, e.g. store the pixel values in Kelvin*100 which gives you a max temp of around 300C.

csedwar, 2020-09-30:
Yeah adding geotags to the tiff is what I was thinking. Perhaps a multiband tiff for each lepton. Frame rates are slow. 1 set in 1 second or so. 

Thanks for the thoughts!