LJMUAstroecology/flirpy #32: TeaxGrabber with ThermalGrabber USB outputs a jambled image

navotoz, 2020-10-02:
Hello,
First , thanks for all the work! Really nice package.

I'm trying to grab images with the _Tau2_ with _ThermalCapture Grabber USB_. 
I cloned the latest version of flirpy (0.2.3), setup a venv and installed using requirements.txt and tried running:
```
from flirpy.camera.tau import TeaxGrabber

camera = TeaxGrabber()
image = camera.grab()
camera.close()
np.save('test', image)

```
The resultent image is scrambled. I think this is sync issue of some sort.
 When I use ThermalCaputre Viewer the output is OK, so I'm positive the camera and grabber are working.

I attached the output of the code. When looking in the ThermalCaputre Viewer it can be seen perfectly.
[test_TeaxGrabber_0.2.3.zip](https://github.com/LJMUAstroecology/flirpy/files/5317650/test_TeaxGrabber_0.2.3.zip)

Thank you again,
with regards,
Navot

jveitchmichaelis, 2020-10-02:
Heya, yes there are sometimes some sync problems. Particularly if you mix commands and streaming. I have some ideas how to solve it, but I think it may occur when the code that looks for the start of a frame goes completely out of phase (e.g. it never finds the start word). But usually it'll work a few times if you unplug/restart - does it work ever? Or does it fall over?

Is also may be a resolution issue? Do you have the 640 or the 320?

navotoz, 2020-10-02:
Thanks for the fast replay.

The resolution is 640x512, and the results are consistent with it.

The magic word is found even without the fix, but I fixed tit anyway. Still no cigar.

I tried restarts/replugging, still no luck..

jveitchmichaelis, 2020-10-02:
Hmm so you're actually getting correct frame sizes, they're just scrambled? I didn't get a chance to look at your file yet, I'll do that when I get a chance. Could it be some config setting on the camera?

navotoz, 2020-10-02:
Regarding the config - I don't know, but the camera does work when using the ThermalCapture software.

jveitchmichaelis, 2020-10-02:
It's possible that TeAx's software handles some extra configuration that Flirpy doesn't. The way their API works is that the camera stream is constantly processed and there's a big switch statement to deal with it (it's a state machine).

To be honest I'm not sure, as we've tested this with two USB Grabbers without issues (beyond the occasional sync error).

Can you send the raw bytes between the two magic "TEAX" words, rather than the converted frame? It looks like the rows in the image are the wrong length.

navotoz, 2020-10-02:
I'm currently just taking pictures of my office.
Attached is the _data_ from `_convert_frame` under `TeaxGrabber`.
[raw_data.zip](https://github.com/LJMUAstroecology/flirpy/files/5319666/raw_data.zip)

jveitchmichaelis, 2020-10-02:
Thanks, I'll see what I can do. So we're expecting images around 20C, which should be word values around 7000.

If you open up the raw here, you can see the sorts of values that are coming back: https://hexed.it/

I believe the row markers are 0x00400040 (NULL @ NULL @). Your data looks fine actually, you can see that there all the words starting with 0xDD are in the correct range (about 7k). So maybe we're just messing up reading the header. The image starts at offset ~14.

jveitchmichaelis, 2020-10-02:
Definitely something odd going on, this is as far as I've got:

```
import re, pickle, numpy as np

r = re.compile(b"\x00\x40")

prev_start, prev_end = None, None
data_slice = b""

with open('../raw_data.pkl', 'rb') as f:
    data = pickle.load(f)

for i in re.finditer(r, data):
    current_start, current_end = i.span()
    
    if prev_end is not None:
        data_slice += data[prev_end+2:current_start]
    
    prev_start = current_start
    prev_end = current_end
    
data = np.frombuffer(data_slice, dtype='>u2')

height = 512
width = 640

buffer = np.zeros((640*512), dtype='>u2')
buffer[:len(data)] = data

imshow(buffer.reshape((height, width)), vmin=6000, vmax=8000)
colorbar()
```

[image]

Something is definitely there, but there does look like something odd going on with the row synchronisation.

jveitchmichaelis, 2020-10-02:
Having a look a the teax source, maybe we just got lucky with the cameras we tested with. It looks like there are some other considerations. I think we're not getting the full frame, and it also looks like those \0x0040 besides being big-endian are PPS values from GPS. Do you have GPS in?

In `tauimagedecoder.cpp`

jveitchmichaelis, 2020-10-02:
Ok success 👍 

``` python
import re, pickle, numpy as np

with open('../raw_data.pkl', 'rb') as f:
    data = pickle.load(f)

"""
Load a single frame
"""
data_slice = b""
prev_start, prev_end = None, None
r = re.compile(b"TEAX")
for i, match in enumerate(re.finditer(r, data)):
    current_start, current_end = match.span()
    
    if prev_end is not None:
        data_slice += data[prev_start:current_start]
        break
    
    prev_start = current_start
    prev_end = current_end
    
data = data_slice

def _convert_frame(data, to_temperature=False, width=336):
    raw_image = np.frombuffer(data[10:], dtype='uint8').reshape((-1,2*(width+2)))
    
    print(raw_image.shape)
    
    raw_image = 0x3FFF & raw_image.view('uint16')[:,1:-1]

    if to_temperature:
        return 0.04*raw_image - 273
    else:
        return raw_image

imshow(_convert_frame(data))
colorbar()
```

[image]

So it looks like your camera is defaulting to a much smaller frame size. I may need to add a command to send to the camera to fix this.

jveitchmichaelis, 2020-10-02:
It would be worth figuring out where this resolution is being set, if indeed you have a 640 model camera and Teax's software gives you the correct resolution. I can't see anything in the Tau software IDD that mentions changing the size of the image that the camera core outputs, it should always be full res. There is a function to window the image (0xED), but I don't see why that would be used.

navotoz, 2020-10-03:
When running
```
from flirpy.camera.tau import TeaxGrabber

camera = TeaxGrabber(width=336, height=256)
image = camera.grab()
camera.close()
np.save('test', image)
```
[This line](https://github.com/LJMUAstroecology/flirpy/blob/b313ef563ca0b58f0a39a58be74f22d3f3011cb2/flirpy/camera/tau.py#L709)  in TeaxGrabber failes, so I added **@staticmethod** before `_convert_frame`.

Anyway, in `_convert_frame` the `frame_width` recived is 338 for some reason, and the image is still scrambled.
I'm attaching all the outputs of the camera. __sync_output_ is the output of `_sync` inside `grab`. _while_loop_output_ is the final output of `grab`, _raw_image_first/second_line_ are the first and second `raw_image`.
[runs.zip](https://github.com/LJMUAstroecology/flirpy/files/5321083/runs.zip)

Interestingly, if I use `width=336`, disregard the `frame_width` from _data_, and write:
`data = data[:-2*512]`
before the _raw_image_ lines, I get the correct image (albit with the smaller resolution).

I'll also write an email to ThermalCapture about setting the resolution in TeaxGrabber.

Thank you.

jveitchmichaelis, 2020-10-03:
Oh I think we're just missing a self

navotoz, 2020-10-03:
> Oh I think we're just missing a self

Yep.

jveitchmichaelis, 2020-10-03:
Almost certainly this is wrong

https://github.com/LJMUAstroecology/flirpy/blob/2605e504da92c396e2071741daa891c548a2929d/flirpy/camera/tau.py#L551

You should change the 2058 to something like (10+4*height)

navotoz, 2020-10-03:
> You should try changing 2058 to something like (10+4*height). Ie account for the 10 bytes in the header plus 4 pad bytes per row.

Yes!! Thank you, works perfectly now.

navotoz, 2020-10-03:
OK, I got it, the camera is actually 336x256. I checked the part number <face palm>.
Well, thank you very much @jveitchmichaelis. It works perfectly now.

closed 2020-10-03 by navotoz

jveitchmichaelis, 2020-10-03:
OK let me push this and I'll bump the version on pypi for you