LJMUAstroecology/flirpy #33: Use case for raw2temp

csedwar, 2020-10-07:
I'm wondering if you can provide an example use case of raw2temp...information online is scarce related to converting boson data to temperatures.

jveitchmichaelis, 2020-10-07:
The main use case is converting FLIR raw formats (eg SEQ from the Duo Pro R) which provide counts and calibration coefficients. This is not possible for the Boson unless you perform a radiometric calibration yourself and compute those coefficients, unfortunately. FLIR has said for the last two years that a radiometric variant is imminent, but nothing has been released yet.

csedwar, 2020-10-07:
Yeah.  That's what I was concerned about. I have 2 calibration targets (v-groove, painted) that are instrumented which I can use to derive the calibration on a per-scene basis, which overall is actually fine...its just annoying.  I'm pretty confident the cores in a boson have no physical reason why one cant do radiometry. The big problem for me is that drift you were talking about if the camera temp changes dramatically during the scene acquisition. Good news for me is my acquisitions are short and i'd wait until the camera has mostly stabilized.

csedwar, 2020-10-07:
Side note (i also have a Duo Pro R), have you had any success dealing with a non-uniformity in temperature across the image.  I see a ~1K delta from center to side looking at my cal-targets. It really appears to be a lens based effect, and probably needs a different NUC or other but i've never been able to figure out if this is actually possible to perform with the duo pro r whereas its super obvious how to do this with the Boson (though not that it seems to really mean that much).

jveitchmichaelis, 2020-10-07:
Hmm we have seen significant vignetting too. One approach I've used for aerial imagery is to just do a sort of dynamic flat field, using the median of the previous 100 frames. That works reasonably well for images captured with motion.

But I don't think you can do much with the Duo itself, it's frustratingly locked down. You'd have to do it in post.

csedwar, 2020-10-07:
Seems like we're learning the same things! Sigh. I really do not understand why the Duo Pro R has fewer capabilities than the Boson in terms of calibration but the Boson cannot provide a linear output. And yeah. a median is likely to work well.  

Do you find the Boson is just a shift (e.g. all the counts move up by the same few hundred counts with the temp of the FPA?). I mean i'm not against picking a reference image and tying everything to that either (when rastering a scene)