LJMUAstroecology/flirpy #34: Tau2 with TeaxGrabber - timeouts in fpa and housing temperatures

navotoz, 2020-10-14:
Hello, 

I'm using Tau2 with the TeaxGrabber.

get_fpa_temperatures() and get_housing_temperatures() sometimes returns the correct values, and sometimes returns errors such as:

Initial packet byte incorrect. Byte was: 67
Error reply from camera. Try re-sending command, or check parameters.

Initial packet byte incorrect. Byte was: 93
Error reply from camera. Try re-sending command, or check parameters.

Initial packet byte incorrect. Byte was: 0
Error reply from camera. Try re-sending command, or check parameters.
.....

I noticed no pattern - sometimes I'm able to get 5 measurements in a row, sometimes it fails after just 1 measurement.
I tried adding sleep() between sending to receiving, without results.
I tried reconnecting the Tau2, and resetting the PC a couple of times.

Thanks for the great package (again :-))

https://github.com/LJMUAstroecology/flirpy/blob/b48b6434fa1572d833c9c42fd57577983dce903f/flirpy/camera/tau.py#L153-L176

jveitchmichaelis, 2020-10-14:
Yep this is currently a bit buggy due to the way the serial stream is implemented. I think it's just losing sync when the messages are sent back from the camera. Not sure if there's an easy fix here without re-writing the stream handling code to be more like a proper state machine.

But we can do the same as before - if you can dump the serial buffer just before the error I can take a look and see if we can figure something out :)

jveitchmichaelis, 2020-10-14:
As before, we can try changing `data = self._read()` to `data += self._read().

One concern is perhaps we accidentally discard the data before waiting for it.

navotoz, 2020-10-14:
I changed to data += self._read() on both functions, without results.

Here are the logs:
[log_first_fail.txt](https://github.com/LJMUAstroecology/flirpy/files/5376310/log_first_fail.txt)
[log_fourth_fail.txt](https://github.com/LJMUAstroecology/flirpy/files/5376312/log_fourth_fail.txt)

jveitchmichaelis, 2020-10-14:
Can you save the entire data buffer from _sync_uart?

navotoz, 2020-10-14:
[log__sync_uart_failed_2.txt](https://github.com/LJMUAstroecology/flirpy/files/5376333/log__sync_uart_failed_2.txt)
[log__sync_uart_failed_7.txt](https://github.com/LJMUAstroecology/flirpy/files/5376334/log__sync_uart_failed_7.txt)

jveitchmichaelis, 2020-10-14:
Try:

     np.save("dump.dat", data)

just before:

     return data[data.find(magic):] 

But this is interesting, it looks like each byte is packed. Ok should be quite easy to fix this, probably needs a new function that will read out a certain number of UART bytes.

```
UART\x01nUART\x01\x00UART\x01\x00UART\x01 UART\x01\x00UART\x01\x02UART\x01yUART\x01?UART\x01\x0cUART\x01IUART\x01\x9cUART
```

navotoz, 2020-10-14:
[dump.zip](https://github.com/LJMUAstroecology/flirpy/files/5376385/dump.zip)

navotoz, 2020-10-14:
[dump_fail.zip](https://github.com/LJMUAstroecology/flirpy/files/5376397/dump_fail.zip)
[dump_success.zip](https://github.com/LJMUAstroecology/flirpy/files/5376398/dump_success.zip)

navotoz, 2020-10-15:
I tried using pickle instead:
```
import pickle
with open("dump.pkl", "wb") as fp:
    dump(data, fp)
```
Results:
[dumps_pickle.zip](https://github.com/LJMUAstroecology/flirpy/files/5382476/dumps_pickle.zip)