LJMUAstroecology/flirpy #36: Got flirpy to work just fine except no color

pyobox, 2020-11-11:
Im running flirpy on my raspberry pi using the example.code and a lepton pure thermal 2.  Worka fine except the picture is just black and white.  How do I add false color?
Here is my code:
``
    Import cv2
    from flirpy.camera.lepton import Lepton
    import numpy as np

    with Lepton() as camera:
        while True:
            img = camera.grab().astype(np.float32)

            # Rescale to 8 bit
            img = 255* (img - img.min())/(img.max()- 
     img.min())

            cv2.imshow('Lepton', img.astype(np.uint8))
            if cv2.waitKey(1) == 27:
                break  # esc to quit
       
    cv2.destroyAllWindows()
``

jveitchmichaelis, 2020-11-11:
Take a look at `cv.applyColorMap`, something like:

```
with Lepton() as camera:
    while True:
        img = camera.grab().astype(np.float32)

        # Rescale to 8 bit
        img = 255* (img - img.min())/(img.max()- 
 img.min())

        img_col = cv2.applyColorMap(img.astype(np.uint8), cv2.COLORMA_INFERNO)

        cv2.imshow('Lepton', img_col)
        if cv2.waitKey(1) == 27:
            break  # esc to quit
   
cv2.destroyAllWindows()
```

There's a tutorial here: https://www.learnopencv.com/applycolormap-for-pseudocoloring-in-opencv-c-python/. For thermal images, you're probably expecting COLORMAP_MAGMA or COLORMAP_INFERNO.

jveitchmichaelis, 2020-11-11:
If you're just using matplotlib then just apply a colormap as normal when you call imshow e.g. `plt.imshow(img, cmap='magma')`

pyobox, 2020-11-11:
I applied the changes to my code that you proposed and I'm getting this error:

img_col = cv2.applyColorMap(img.astype(np.uint8), cv2.COLORMA_INFERNO)
AttributeError: module 'cv2.cv2' has no attribute 'COLORMA_INFERNO'
>>> 

Btw thank you so much for the quick response

jveitchmichaelis, 2020-11-11:
oops typo - `cv2.COLORMAP_INFERNO`

pyobox, 2020-11-11:
Changed it to COLORMAP_JET works like a charm thank you so much!

closed 2020-11-11 by pyobox

jveitchmichaelis, 2020-11-11:
No problem. If you can't get INFERNO to work (that's the one that looks most like FLIR's standard "iron" colormap) then you can try running `pip install --upgrade opencv-python-headless` (you need version > 4.1 I think)

pyobox, 2020-11-12:
A little update:  I've been using an older version of openCv it was like 3.x.x because that's the only way to get opencv cascade commands to work, but i upgraded it back to 4.1.0.25 and all three Color Maps work perfectly! Again thanks for the help!