LJMUAstroecology/flirpy #37: camera.boson - AttributeError: type object 'Boson' has no attribute 'logger'

helarsen, 2021-01-25:
I get the following error when attaching to a Boson camera:   
> File "C:\Program Files\Python37\lib\site-packages\flirpy\camera\boson.py", line 146, in find_video_device
    self.logger.info("Device ID: {}".format(device_id))
AttributeError: type object 'Boson' has no attribute 'logger'

The reason is that the         
>@classmethod
    def find_video_device(self):

does not have access to the _self_ object where the logger is instantiated as _self.logger = logging.getLogger(__name__)_

Repair:
If the @classmethod is removed, the code works in my particular case. I am not able to say if this is the best or correct modification.

Thanks for providing the library and great if someone can update the code.

helarsen, 2021-01-25:
Edit: I see that this issue has been corrected in the latest code on the repository. I installed using 
>pip install flirpy

which currently pulls the code (0.2.3) from https://pypi.org/search/?q=flirpi - So an update is needed there.

jveitchmichaelis, 2021-01-26:
Thanks for the nudge, I'll do a version bump shortly.

closed 2021-01-26 by jveitchmichaelis