LJMUAstroecology/flirpy #38: Error "'split_file' is not defined" while calling flirpy.io.teax.process_file()

uydtm, 2021-01-25:
Thank you for developing this package! It really is very useful.

I want to convert .TMC images to more common file formats (such as .tiff or .jpg). While trying to use the section `flirpy.io.teax` for this purpose, I came across an issue which I cannot seem to resolve. No matter how I try calling the function  `process_file()` , I always receive an Error message: `NameError: name 'split_file' is not defined`. 

Has anyone else ever encountered a similar problem?

jveitchmichaelis, 2021-01-26:
TMC files are currently not supported, for now you need to use TeAx's software. I was able to partially reverse engineer the file format, but not to a point where it would be useful, unfortunately.

uydtm, 2021-01-26:
Hello Mr. Veitch-Michaelis,

Thank you so much for your quick response and clarification! I’m sorry to hear it isn’t possible.
Maybe it would be best to change the package description to exclude tmc where it says “Convert FLIR file formats (e.g. seq, fff, tmc, tfc) to geotagged readable images”? Then no one else will get confused as I did. I will add a comment to the issue I posted yesterday.

Kind regards,
Elena Vollmer

uydtm, 2021-01-26:
Thank you so much for your quick response and clarification! I’m sorry to hear it isn’t possible. 
Maybe it would be best then to change the package description to exclude tmc where it says “Convert FLIR file formats (e.g. seq, fff, tmc, tfc) to geotagged readable images”?

jveitchmichaelis, 2021-01-26:
https://github.com/LJMUAstroecology/flirpy/blob/master/README.md

Done! Sorry about that

closed 2021-01-26 by jveitchmichaelis