LJMUAstroecology/flirpy #39: Multiple Boson under win32 - find_cameras returns only first

helarsen, 2021-01-28:
I have several BOSON on Windows 10 system. Auto detection seems not to work.
"find_cameras.exe" returns only one camera.
Is that correctly observed?

If so, I would like to suggest that at more appropriate function would be to return a list of all devices.

jveitchmichaelis, 2021-01-29:
Yes, currently the script only returns one camera. You can take a look at the script: https://github.com/LJMUAstroecology/flirpy/blob/master/find_cameras/find_cameras/find_cameras.cpp

Just have it print out every camera rather than just the first. Feel free to submit a pull request!

closed 2021-01-29 by jveitchmichaelis

helarsen, 2021-01-29:
thanks. And sorry I should have seen #24. I will investigate if I can come up with a solution. The enum com devices is easy - done that already. The cross linking to video device is the hard part.

jveitchmichaelis, 2021-01-29:
Yeah, I would suggest just returning a vector of IDs and then printing them comma separated (or something similar). Unfortunately I no longer have multiple Bosons to test with otherwise I'd have a go - though I can update the code and if you're willing to test it we can work together on it. The difficult bit is then matching the serials, but I think that should be possible.

helarsen, 2021-01-29:
I am not much familier with this workflow.
The code below is tested and will return a list of COM ports and the Boson serial number.
The list could also be returned as [["COM35", 12456], ["COM8",23456]] to keep items as a camera property.
Anyways - this code does not break much ground.
I would be happy to assist and can test with at least 2 Bosons.
henning

>    @classmethod
    def get_serial_device_list(self):
       """
        Attempts to find the serial ports that has a Boson connection and returns a list of COM port and serial number
        Returns
        -------
            list of string
                serial port name
            list of int
                Boson serial number
        """
          port = None
          device_list = list_ports.comports()
          device_list = sorted(device_list, key=lambda dev: dev.device) # in case os uses random order.
          VID = 0x09CB
          PID = 0x4007
          ports = [ device.device for device   in device_list if device.vid == VID and device.pid == PID]
          sns = [ int(device.serial_number) for device   in device_list if device.vid == VID and device.pid == PID]
          return ports, sns

jveitchmichaelis, 2021-01-29:
So if you can extract the serial number from DirectShow as well, then that should give you the correct device. Could you check whether the serial number you can get from the device matches any of the properties listed by Windows (any unique matching parameter would do actually)? You can check this by going to Device Manager and checking the Details tab for the camera there (you should be able to see e.g. the PID, VID and friendly name, etc).

helarsen, 2021-01-29:
One device seen by device manager:
[image]

Second device seen by device manager:
[image]

So there is a common value seen in both views. One can get this value via COM listing in python (code above with minor modification):
for COM8:
location = '1-1.2:x.2' or
hwid='USB VID:PID=09CB:4007 SER=85786 LOCATION=1-1.2:x.2'
for COM35
location = '1-1.3.2:x.2'
hwid='USB VID:PID=09CB:4007 SER=53112 LOCATION=1-1.3.2:x.2'

Although a different format it is clearly the same parameter and thus a link between the two views. BUT
**How to get the video location information programmatically?**

jveitchmichaelis, 2021-01-30:
You should be able to query that using the cpp script, in the same way we check for the "friendly name" e..g

```
hr = pPropBag->Read(L"Location information", &var, 0);
```

helarsen, 2021-01-30:
Unfortunately pPropBag interface calls
>hr = pPropBag->Read(L"Location Information", &var, 0);
hr = pPropBag->Read(L"Location information", &var, 0);
hr = pPropBag->Read(L"location", &var, 0);
hr = pPropBag->Read(L"Location", &var, 0);

does not return any value to var. But

>hr = pPropBag->Read(L"DevicePath", &var, 0);

returns 
>var=
>BSTR = 0x01203734 L"\\\\?\\usb#vid_09cb&pid_4007&mi_00#d&2b4821fa&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\\global"

and for the second camera
>var=
BSTR = 0x01203734 L"\\\\?\\usb#vid_09cb&pid_4007&mi_00#3&26014a8a&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\\global"

As seen slightly different but does not correlate well to what serial port enumeration gives.
So back to the guesswork - I wish I could find documentation for this horrible pPropBag interface

jveitchmichaelis, 2021-01-31:
Yeah, I've not been able to find a way of enumerating valid keys for an IPropertyBag. This might be one for Stack Overflow to answer. I'm not super familiar with COM programming.

helarsen, 2021-01-31:
[https://docs.microsoft.com/en-us/windows/win32/directshow/selecting-a-capture-device](url)
says

1. The "FriendlyName" and "Description" properties are suitable for displaying in a UI. The "FriendlyName" property is available for every device. It contains a human-readable name for the device.
3. The "Description" property is available only for DV and D-VHS/MPEG camcorder devices. For more information, see MSDV Driver and MSTape Driver. If available, it contains a description of the device which is more specific than the "FriendlyName" property. Typically it includes the vendor name.
4. The "DevicePath" property is not a human-readable string, but is guaranteed to be unique for each video capture device on the system. You can use this property to distinguish between two or more instances of the same model of device.

so basically there are only Firendlyname and devicePath to read that way. DevicePath is unique (example is in my previous post) but it does not help here due to lack of visibility via the serial port.

Tried to switch the device into ramp mode and hoped that way to generate an image which is easy to identify. This has for the moment stranded on the fact that the test ramps I get are not fixed patterns and are not reproducible which makes it hard to use. I am however confident that it can be done that way.
Problem with test ramp is that it is (also) poorly documented and there are a surprisingly many types of ramps not only patterns but also the location they are at.
**Does anyone reading here have experience with setting test ramps?**

jveitchmichaelis, 2021-01-31:
Why not just set `FLR_TESTRAMP_VTST_CHECKERBOARD` (in the Software IDD) - these should be fixed? What image do you get out?

jveitchmichaelis, 2021-02-01:
Actually, did you try removing spaces? The following works for me with an integrated camera:

`hr = pPropBag->Read(L"LocationInformation", &var, 0);`

jveitchmichaelis, 2021-02-01:
This example compiles fine for me on Windows 10 and does print out the location correctly. Still need to match with the correct DShow order though:  https://stackoverflow.com/questions/3438366/setupdigetdeviceproperty-usage-example

helarsen, 2021-02-01:
Great work!
I think its almost in the PropBag!

This is what I get looping for all flir:
[image]

and it matches what device manager show.
Remains to correlate 
0009.0000.0000.**001.002.000**.000.000.000 with location = '1-**1.2**:x.2'
0009.0000.0000.**001.003.002**.000.000.000 with location = '1-**1.3.2**:x.2'
because the COM query gives
for COM8:
location = '1-1.2:x.2' or
hwid='USB VID:PID=09CB:4007 SER=85786 LOCATION=1-1.2:x.2'
for COM35
location = '1-1.3.2:x.2'
hwid='USB VID:PID=09CB:4007 SER=53112 LOCATION=1-1.3.2:x.2'

Fortunately this can and should be done in python.
Guessing that it is a matter of comparing the fields in bold and setting .000 = "" and hoping the format is more or less fixed.

helarsen, 2021-02-01:
> Did you try removing spaces with the query string? The following works for me with an integrated camera:
> 
> `hr = pPropBag->Read(L"LocationInformation", &var, 0);`

no did not work in this context

jveitchmichaelis, 2021-02-01:
> 
> 
> > Did you try removing spaces with the query string? The following works for me with an integrated camera:
> > `hr = pPropBag->Read(L"LocationInformation", &var, 0);`
> 
> no did not work in this context

Yeah I wondered if that was an easy fix, but then I realised I hadn't checked the output was actually a success and I was getting random strings every time.

jveitchmichaelis, 2021-02-01:
For now let's assume that it's those three fixed fields. If someone else has an issue with it, then we can worry about it later!

helarsen, 2021-02-01:
I have implemented the algorithms to query the video and the serial port and match up giving a list of cameras:
>[{"vid":1, "port":"COM3", "sn":1234}, {"vid":2, "port":"COM45", "sn":1256} ], is_race (note 1)

There remain a rather important work incorporating this into the original class. This require restructuring of the Boson class.
This is a work which is best done by the main author as it affects the architecture.

One odd thing I see (I may say an asymmetry)  with the current architecture is that when the class is created, only the com port is opened and not the associated video port. I think they should both be opened in the constructor so that the camera is ready for any type of operation (cap is initialized). With the new feature this should be possible to do in a consistent manner.

Also one should decide how to identify a camera to the constructor: By the vid, the port or maybe by the serial number. Again a fundamental change to the architecture.

(note 1) One note about the list of cameras: If there is a PnP event between the two type of device queries, the procedure will fail. The reason is that the vid always starts at the lowest value, so if we have vid 1 and 2 and 1 gets unplugged, the remaining will change to vid=1
To allow handling of this, the code is checking that the two queries have the same number of devices and if not the second argument returned is set to True meaning "RaceCondition"

Let me know any thoughts about further progress and suggestions.

jveitchmichaelis, 2021-02-04:
I'll take a look at this properly shortly, but I would suggest that rather returning a flag for a race condition, just make the whole thing error and ask the user to try again. This seems like a very narrow edge case (if you pull a camera during enumeration?) and it would be simpler to just leave it as an assert somewhere.

helarsen, 2021-02-05:
ok I see 4 options to signal a pnp event during the execution of the two device enumerations which causes inconsistency.

1. ignore and return whatever camera configuration was found. Not good idea as it masks the error.
2. return None
3. return a flag (race=True) like i describe and let the caller deal with it
4. throw an error of some kind inside the enumerator

I really have no præference of the 3

jveitchmichaelis, 2021-02-05:
There is also option 5 which is to repeat the check until (for say 3 retries) the results of the enumeration are consistent?