LJMUAstroecology/flirpy #44: Lepton 3.5, PurethermalMini,Capture issue

basavarajamogi, 2021-02-09:
I have following error in Python capture using Purethermal mini
TypeError: Argument 'value' must be double, not bool

jveitchmichaelis, 2021-02-09:
This should be fixed already https://github.com/LJMUAstroecology/flirpy/commit/a094fdeb09c6f775fd82e7f0287b461ab792382d

Did you try pulling from the repository directly?

basavarajamogi, 2021-02-10:
> This should be fixed already a094fde
>
> Did you try pulling from the repository directly?

Thank you Josh. I directly pulled the repository but the Raspberry pi is
showing 'skbuild': No module found error

jveitchmichaelis, 2021-02-10:
That's probably an issue trying to build opencv. If you can't install it as a dependency, try and install it separately. Have a look at:

https://stackoverflow.com/questions/63448467/installing-opencv-fails-because-it-cannot-find-skbuild

basavarajamogi, 2021-02-12:
Thank you Josh. I solved that issue.
But I am still not able to grab the picture. It gives the following error.

  File
"/usr/local/lib/python3.7/dist-packages/flirpy-0.2.3-py3.7.egg/flirpy/camera/lepton.py",
line 156, in grab
    self.setup_video(device_id)
  File
"/usr/local/lib/python3.7/dist-packages/flirpy-0.2.3-py3.7.egg/flirpy/camera/lepton.py",
line 97, in setup_video
    raise ValueError("Lepton not connected.")
ValueError: Lepton not connected.

When I check lsusb it gives in a list   "Bus 001 Device 004: ID 1e4e:0100
Cubeternet Webcam".

basavarajamogi, 2021-02-12:
Okay, Now everything is solved. The problem was with PT mini board. When I checked with another PT Mini board everything started working fine.
Only last issue I have is following.
I am getting raw image matrix but with following warning.
[ WARN:0] global /tmp/pip-wheel-qd18ncao/opencv-python/opencv/modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index

How to get over with this warning?

jveitchmichaelis, 2021-02-13:
Check that the Lepton is fully seated on your first board - it's quite easy for the camera module to come loose.

This can happen sometimes with OpenCV. If you have no other cameras connected, you can try forcing the camera index to be `-1`.

basavarajamogi, 2021-03-26:
I got everything working but sometimes I get following error. When I restart the raspberry pi after sometime, it starts working and after few hours or something throws the same error.
Can you please suggest me something?
[ WARN:0] global /tmp/pip-wheel-qd18ncao/opencv-python/opencv/modules/videoio/src/cap_v4l.cpp (1004) tryIoctl VIDEOIO(V4L2:/dev/video1): select() timeout.
[ WARN:0] global /tmp/pip-wheel-qd18ncao/opencv-python/opencv/modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index

jveitchmichaelis, 2021-03-28:
This should probably be a question for OpenCV as that's what's being used to open the camera.

basavarajamogi, 2021-03-28:
Thank you I will check out there.

jveitchmichaelis, 2023-07-04:
Closing as this seems to be a non-Flirpy issue - hope you got it fixed!

closed 2023-07-04 by jveitchmichaelis