LJMUAstroecology/flirpy #45: How to read fff file without TypeError?

Ikhwansong, 2021-02-28:
Hi thanks for your devotion to this wonderful library flirpy

I ran the code split_seqs to convert seq file to many fffs.
After processing, when i read one fff file with the function Fff that is imported by from _flirpy.io.fff import Fff_.
Below TypeError occurs

TypeError: __init__() got an unexpected keyword argument 'height'

What should i do?

jveitchmichaelis, 2021-03-01:
In the newer versions of the code height/width were removed, so just try a = Fff("frame_000000.fff") - flirpy should be able to determine the image shape directly now.

Ikhwansong, 2021-03-01:
Thanks for the rapid answer.

I followed your advise, but i suffered another Error.

My code
from flirpy.io.fff import Fff
a = Fff('./test/frame_000000.fff')

Traceback (most recent call last):
  File "flir_imshow_test.py", line 11, in <module>
    a = Fff('./test/frame_000000.fff')
  File "/home/ihsong/Documents/project/flirpy-master/flirpy/io/fff.py", line 28, in __init__
    self._find_data_offset()
  File "/home/ihsong/Documents/project/flirpy-master/flirpy/io/fff.py", line 115, in _find_data_offset
    res = s.unpack_from(self.data, record_offset)
struct.error: unpack_from requires a buffer of at least 32 bytes

Help me...

jveitchmichaelis, 2021-03-01:
Yes, this is partially the reason why this isn't on PyPI yet. Can you try with a version of flirpy installed using `pip`?

Ikhwansong, 2021-03-01:
I made two anaconda env(the one run in the clone directly, the other install the flirpy by pip)

I met a new Error.

My code
from flirpy.io.fff import Fff
a = Fff('./test/frame_000000.fff')

Traceback (most recent call last):
  File "/home/ihsong/Documents/project/flirpy-master/flir_imshow_test.py", line 11, in <module>
    a = Fff('/home/ihsong/Documents/proejct/flirpy-master/yp/raw/frame_000001.fff')
  File "/home/ihsong/Documents/project/flirpy-master/flirpy/io/fff.py", line 28, in __init__
    self._find_data_offset()
  File "/home/ihsong/Documents/project/flirpy-master/flirpy/io/fff.py", line 115, in _find_data_offset
    res = s.unpack_from(self.data, record_offset)
struct.error: unpack_from requires a buffer of at least 1073741856 bytes for unpacking 32 bytes at offset 1073741824 (actual buffer size is 617036)

jveitchmichaelis, 2021-03-01:
Hmm ok can you try with Exiftool directly? This is a known problem for some cameras.

e.g. `exiftool ./test/frame_000000.fff -flir:all -H -G`

Ikhwansong, 2021-03-01:
Thanks for your help.
Actually, there was my one mistake that after installing by the pip, i didn't run the code at the external directory(i.e., i ran the code in the clone)

So, i moved the external directory and executed the same code, then it worked!
If i meet another problem, i will be back to question.

Thanks again!

closed 2021-03-01 by Ikhwansong