LJMUAstroecology/flirpy #46: Boson: Disable all Non-Uniformity Correction

vishwa91, 2021-03-06:
First, thanks a ton for the wonderful codebase -- made my life so much easier.

I wanted to know if it is possible to disable all corrections on the Boson 640 camera? I want to get raw data for performing custom NUC and hence the question.

Thank you.

jveitchmichaelis, 2021-03-06:
Hiya, this is probably possible. You'd likely need multiple commands because there are several stages of correction that the camera performs by default (and I'm not sure which you'd want to turn off for your application). It's a matter of implementing the commands defined in the Boson Software idd, but that's fairly easy if you know which. We'd need to add the relevant commands to the Boson class - ultimately it'd be good to implement the entire camera interface at some point...

vishwa91, 2021-03-06:
I agree with the multi-step command, as the GUI requires switching off several options. I will look into the Boson commands reference and try them to this codebase. Thanks!

jveitchmichaelis, 2021-03-06:
Yes, each command should be a separate API call. If you get them working, please consider submitting a pull request (ideally also with a test case for each command- so you should write a set/getfunction for each). Otherwise if you have any questions about implementation, feel free to ask.

vishwa91, 2021-03-06:
I will follow the same template as the code base, and add test functions. Once I test on the Boson camera I have, I will submit a pull request. Thanks for the support!

nrwiederhold, 2022-07-06:
What is the status of this issue? Do these options exist in the code base now?

jveitchmichaelis, 2022-07-06:
No PR was submitted, but you're welcome to add it - it should be quite easy to do, but I don't have access to a Boson at the moment.

socome, 2023-01-04:
@jveitchmichaelis Could you teach me how to turn off the NUC on boson camera .? I will try and PR about that. I want to turn off the NUC because of image block when NUC is running

jveitchmichaelis, 2023-01-04:
Do you mean flat fielding? This causes a slight pause when the camera takes a calibration image. This is different to NUC.

socome, 2023-01-04:
@jveitchmichaelis Oh .. When i read [this post](https://www.flir.com/discover/professional-tools/what-is-a-non-uniformity-correction-nuc/), i think NUC is main reason of thermal image sometimes freezes and the camera makes a clicking noise. However, do you mean this problem caused by FFC..? I will try to use the "set_ffc_manual" function. Thank you!

jveitchmichaelis, 2023-01-04:
Ah I think they use it interchangeably. From the article:

>  It is also referred to as an FFC (flat field correction

Usually I’ve seen NUC used to refer to the various other settings the camera uses to correct the image (see the beginning of this thread).