LJMUAstroecology/flirpy #48: Multiple Flir Lepton Image Streaming

Ledroz, 2021-04-05:
Hi,

I am having trouble streaming 2 streams of images from 2 flir lepton 3.5s connected via the Purethermal 2 USB boards connected to windows pc.
I'm kinda new to programming and python as well.
I intend to get the image array via flirpy->colour with matplotlib cmap-> read frames with opencv
I tried reading directly from opencv but the image colouring is not as good as through matplotlib

I was able to stream 1 image with 1 flir, but I'm unsure how i can do it for 2 or more cameras.

I wrote this simple code to stream 2 images:

from flirpy.camera.lepton import Lepton
from matplotlib import pyplot as plt
import numpy as np
import cv2

camera0 = Lepton(1)
camera1 = Lepton(2)

image0 = camera0.grab()
image1 = camera1.grab()

# print(image0)
plt.imshow(image0, cmap="inferno" )#, interpolation='nearest'
plt.show()

# print(image1)
plt.imshow(image1, cmap="inferno" )#, interpolation='nearest'
plt.show()

camera0.close()
camera1.close()

jveitchmichaelis, 2021-04-05:
Hi,

 

There shouldn’t be any reason why you can’t open two cameras like you do in your code. Do you get some kind of error message? Provided the indices of your cameras are correct, your code should work.

 

Check in device manager that both devices are actually being detected as cameras, perhaps?

 

Thanks!

Ledroz, 2021-04-06:
Hi,

Thank you for writing back! The devices are detected in device manager. I see two purethermal boards and I can open them normally via the windows camera app.

I modified the code and it works now:

# Libraries
from flirpy.camera.lepton import Lepton
import numpy as np
import cv2

# Functions
def stackImages(scale,imgArray):
    rows = len(imgArray)
    cols = len(imgArray[0])
    rowsAvailable = isinstance(imgArray[0], list)
    width = imgArray[0][0].shape[1]
    height = imgArray[0][0].shape[0]
    if rowsAvailable:
        for x in range ( 0, rows):
            for y in range(0, cols):
                if imgArray[x][y].shape[:2] == imgArray[0][0].shape [:2]:
                    imgArray[x][y] = cv2.resize(imgArray[x][y], (0, 0), None, scale, scale)
                else:
                    imgArray[x][y] = cv2.resize(imgArray[x][y], (imgArray[0][0].shape[1], imgArray[0][0].shape[0]), None, scale, scale)
                if len(imgArray[x][y].shape) == 2: imgArray[x][y]= cv2.cvtColor( imgArray[x][y], cv2.COLOR_GRAY2BGR)
        imageBlank = np.zeros((height, width, 3), np.uint8)
        hor = [imageBlank]*rows
        hor_con = [imageBlank]*rows
        for x in range(0, rows):
            hor[x] = np.hstack(imgArray[x])
        ver = np.vstack(hor)
    else:
        for x in range(0, rows):
            if imgArray[x].shape[:2] == imgArray[0].shape[:2]:
                imgArray[x] = cv2.resize(imgArray[x], (0, 0), None, scale, scale)
            else:
                imgArray[x] = cv2.resize(imgArray[x], (imgArray[0].shape[1], imgArray[0].shape[0]), None,scale, scale)
            if len(imgArray[x].shape) == 2: imgArray[x] = cv2.cvtColor(imgArray[x], cv2.COLOR_GRAY2BGR)
        hor= np.hstack(imgArray)
        ver = hor
    return ver

# Camera Calls
cam1 = Lepton()
cam2 = Lepton()

while True:
    # Get images
    img1 = cam1.grab(0).astype(np.float32) #.grab() takes serial port no. aka camera ID
    img2 = cam2.grab(3).astype(np.float32)

    # Linear AGC
    img1 = 255 * (img1 - img1.min()) / (img1.max() - img1.min())
    img2 = 255 * (img2 - img2.min()) / (img2.max() - img2.min())

    # Colourise
    img_col1 = cv2.applyColorMap(img1.astype(np.uint8), cv2.COLORMAP_BONE)
    img_col2 = cv2.applyColorMap(img2.astype(np.uint8), cv2.COLORMAP_BONE)

    # Enlarge
    img_col1 = cv2.resize(img_col1, (640, 480), interpolation=cv2.INTER_CUBIC)
    img_col2 = cv2.resize(img_col2, (640, 480), interpolation=cv2.INTER_CUBIC)

    # Rotate
    # img1 is for right camera
    (h, w) = img_col1.shape[:2]
    (cX, cY) = (w // 2, h // 2)
    M = cv2.getRotationMatrix2D((cX, cY), 90, 1.0)  # rotates 90 degrees
    img_col1 = cv2.warpAffine(img_col1, M, (w, h))
    # img2 is for left camera
    (h, w) = img_col2.shape[:2]
    (cX, cY) = (w // 2, h // 2)
    M = cv2.getRotationMatrix2D((cX, cY), -90, 1.0)  # rotates 90 degrees
    img_col2 = cv2.warpAffine(img_col2, M, (w, h))

    # Display images
    imgStack = stackImages(1,([img_col2, img_col1]))
    cv2.imshow('Stereo', imgStack)

    if cv2.waitKey(1) == 27:
        break

cam1.close()
cam2.close()

closed 2021-05-07 by jveitchmichaelis