LJMUAstroecology/flirpy #49: "pip install flirpy" breaks previously installed OpenCV version 3.4

ichu-lsa, 2021-05-05:
After running "pip install flirpy" I get an error from cv2.imshow() saying:

The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'

This was previously working before the pip install. I'm using OpenCV 3.4

ichu-lsa, 2021-05-05:
It seems like it's switched my version to OpenCV 4.5.1 somehow. Is it intended that we use this specific version of OpenCV?

ichu-lsa, 2021-05-05:
Collecting flirpy
  Downloading https://files.pythonhosted.org/packages/e5/6a/e1ec347cb1643342b250e6196321233088fba926f07ce0323bc2cf92dbe9/flirpy-0.2.3-py2.py3-none-any.whl (10.2MB)
     |████████████████████████████████| 10.2MB 6.4MB/s
Collecting psutil<6.0.0,>=5.6.7 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/71/ce/35107e81e7eae55c847313f872d4258a71d2640fa04f57c5520fc81473ce/psutil-5.8.0-cp37-cp37m-win_amd64.whl (244kB)
     |████████████████████████████████| 245kB 6.8MB/s
Collecting pyserial<4.0,>=3.4 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90kB)
     |████████████████████████████████| 92kB ...
Collecting pyftdi<0.52.0,>=0.51.2 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/80/9e/0893cd66e60946b881651012bbbb673a6bac1ceddf71d196fa17fb09dffa/pyftdi-0.51.2-py3-none-any.whl (150kB)
     |████████████████████████████████| 153kB ...
Collecting opencv-python-headless<5.0,>=4.0 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/7f/4d/4a6353f32552be7f36c0e7179275718d30c3287c9c4ab4d1ba99e26a0948/opencv_python_headless-4.5.1.48-cp37-cp37m-win_amd64.whl (34.8MB)
     |████████████████████████████████| 34.8MB 6.8MB/s
Collecting pyudev<0.23.0,>=0.22.0 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/72/c8/4660d815a79b1d42c409012aaa10ebd6b07a47529b4cb6880f27a24bd646/pyudev-0.22.0.tar.gz (85kB)
     |████████████████████████████████| 92kB 6.1MB/s
Collecting pyusb<2.0.0,>=1.0.2 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/b5/28/b857ac783257f142932b23379d761a3d9becf6deecf5d14075ec19bdb890/pyusb-1.1.1-py3-none-any.whl (58kB)
     |████████████████████████████████| 61kB ...
Collecting tqdm<5.0,>=4.42 (from flirpy)
  Downloading https://files.pythonhosted.org/packages/72/8a/34efae5cf9924328a8f34eeb2fdaae14c011462d9f0e3fcded48e1266d1c/tqdm-4.60.0-py2.py3-none-any.whl (75kB)
     |████████████████████████████████| 81kB ...
Requirement already satisfied: numpy>=1.14.5 in c:\users\ichu\anaconda3\lib\site-packages (from opencv-python-headless<5.0,>=4.0->flirpy) (1.18.2)
Requirement already satisfied: six in c:\users\ichu\anaconda3\lib\site-packages (from pyudev<0.23.0,>=0.22.0->flirpy) (1.12.0)
Building wheels for collected packages: pyudev
  Building wheel for pyudev (setup.py) ... done
  Stored in directory: C:\Users\ichu\AppData\Local\pip\Cache\wheels\39\fd\ac\f5251065fe762db2ef2c619c9113fb8c96b07ddb37ee85fd63
Successfully built pyudev
ERROR: spyder 3.3.6 requires pyqt5<5.13; python_version >= "3", which is not installed.
ERROR: spyder 3.3.6 requires pyqtwebengine<5.13; python_version >= "3", which is not installed.
Installing collected packages: psutil, pyserial, pyusb, pyftdi, opencv-python-headless, pyudev, tqdm, flirpy
  Found existing installation: psutil 5.6.3
    Uninstalling psutil-5.6.3:
      Successfully uninstalled psutil-5.6.3
  Found existing installation: tqdm 4.32.1
    Uninstalling tqdm-4.32.1:
      Successfully uninstalled tqdm-4.32.1
Successfully installed flirpy-0.2.3 opencv-python-headless-4.5.1.48 psutil-5.8.0 pyftdi-0.51.2 pyserial-3.5 pyudev-0.22.0 pyusb-1.1.1 tqdm-4.60.0

jveitchmichaelis, 2021-05-07:
Yes, because there are certain features that flirpy uses that aren't available in OpenCV < 4. If you need to use OpenCV 3 for a different project I'd recommend installing flirpy inside a virtualenv or a conda environment.

closed 2021-05-07 by jveitchmichaelis