LJMUAstroecology/flirpy #50: AttributeError: 'str' object has no attribute 'decode'

mctosima, 2021-05-20:
Hi, I just running the example code using MacOS 11.3.1, opencv-python 4.5.2.52 (also the headless), python 3.9.5 with PureThermal 2 and Lepton 2.5

```
from flirpy.camera.lepton import Lepton

cam = Lepton()
image = cam.grab()
print(image)
print(cam.frame_count)
print(cam.frame_mean)
print(cam.ffc_temp_k)
print(cam.fpa_temp_k)
cam.close()
```
The error showed `AttributeError: 'str' object has no attribute 'decode'`

Do I miss something ? I already installed all the package required.
I also try to run this code but the result is same

```
import cv2
from flirpy.camera.lepton import Lepton

with Lepton() as camera:
	while True:
		img = camera.grab().astype(np.float32)

		# Rescale to 8 bit
		img = 255*(img - img.min())/(img.max()-img.min())
		
		# Apply colourmap - try COLORMAP_JET if INFERNO doesn't work.
		# You can also try PLASMA or MAGMA
		img_col = cv2.applyColorMap(img.astype(np.uint8), cv2.COLORMAP_INFERNO)

		cv2.imshow('Lepton', img_col)
		if cv2.waitKey(1) == 27:
			break  # esc to quit
		
cv2.destroyAllWindows()

```

Thank's for helping and sharing

jveitchmichaelis, 2021-06-02:
Are you installing from git or from pypi? I think this issue was already corrected (it was due to a double decode), but it may not have reached pip yet.

planter01, 2021-06-28:
I'm getting a similar error, with Windows 10 and a Boson camera:

from flirpy.camera.boson import Boson
camera = Boson()
images = camera.grab()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\Python37\lib\site-packages\flirpy\camera\boson.py", line 256, in grab
    self.setup_video(device_id)
  File "C:\Program Files\Python37\lib\site-packages\flirpy\camera\boson.py", line 208, in setup_video
    device_id = self.find_video_device()
  File "C:\Program Files\Python37\lib\site-packages\flirpy\camera\boson.py", line 146, in find_video_device
    self.logger.info("Device ID: {}".format(device_id))
AttributeError: type object 'Boson' has no attribute 'logger'

-----------------------------------------------
I installed flirpy through Pycharm.  Is there a more up-to-date repository?  Above, Git and PyPi were mentioned, but I didn't see which one had the freshest code.  If there's not an easy fix, will try downgrading from 3.7 to 3.5.  

Advice appreciated.

jveitchmichaelis, 2021-06-28:
@planter01 try installing from git directly. This repository will always have the freshest code. This logging issues is fixed in the main branch.

planter01, 2021-06-28:
That worked.  Thanks very much!

closed 2021-10-02 by jveitchmichaelis