LJMUAstroecology/flirpy #51: How to reduce the buffer size of flir boson

VegetarianLion258, 2021-05-21:
I connect a depth camera and a boson320 at the same time and find that boson always show frame about 1 sec later. I have face the same problem using USB camera, and solve by using usbCam.set(cv2.CAP_PROP_BUFFERSIZE, 1). that means I reduce camera buffer size to 1. I think I can solve this problem same way, but I do not know how to reduce the buffer size. Does someone know how to reduce buffer size? Thanks a lot :-D

VegetarianLion258, 2021-05-22:
well, I find the solution, add
self.cap.set(cv2.CAP_PROP_BUFFERSIZE, 1)
under line 217 (self.cap = cv2.VideoCapture(device_id + cv2.CAP_V4L2)) at boson.py,

jveitchmichaelis, 2021-06-02:
Thanks for solving your problem - this is more of an OpenCV issue than with Flirpy. You could also try the threaded classes in this library.

closed 2021-06-02 by jveitchmichaelis