LJMUAstroecology/flirpy #54: RGB synchronization fails if there are more IR than RGB frames

LukasBommes, 2021-08-19:
I wanted to point out a bug in the split_seqs script. The synchronization between IR and RGB works fine as long as there are more RGB than IR frames. However, if there are more Ir than RGB frames, the synchronization logic fails. 

I ran into this issue after switching from the 8 Hz Flir Duo Pro R to the 30 Hz version. As the visual stream is at 29.87 Hz, there are more IR than RGB frames generated.

jveitchmichaelis, 2021-08-20:
Thanks for this, I'll take a look!

LukasBommes, 2021-08-28:
If you want, I can send you my code in about two weeks (I'm currently on holiday). The logic is the same as in your code, just that I invert everything in case n_IR > n_RGB.

However, even after this fix snychronization is still rather poor. For long sequences (~20k frames) the IR and RGB streams are up to 10 frames off in the middle of the sequence. Also, for my camera (Zenmuse XT2) the IR stream hangs once in a while when the camera performs recalibration.

I was thinking of using the frame timestamps (from EXIF tags) for synchronization. In the TIFF stack (which I use instead of SEQs) each frame has a millisecond-accurate walltime associated to it. For the RGB stream there is only a millisecond-accurate relative timestamp starting from zero. However, I am in doubt about the timestamps of the IR stream as the recalibration procedure does not show up here.

It would be interesting to know whether you are able to synchronize your streams properly. So far, the problem seems quite tough. I was even thinking of extracting descriptors from both IR and RGB stream and matching descriptors of each IR frame to temporally neighbouring RGB streams. Something like this: https://la.disneyresearch.com/publication/actionsnapping/ The main difficulty is that we have two different modalities, which makes typical feature descriptors, such as ORB, SIFT, and Bag-of-words unsuitable.

jveitchmichaelis, 2021-08-29:
Sure! This should be a simple fix and it's on my radar anyway. Sync is very
difficult, primarily because there is (a) a non-zero start offset between
the two streams (b) pretty much all the cameras break synchronisation when
they flat-field (I guess we could detect this by detecting static IR
frames?). In practice I've had to do this manually most of the time. I
tried a few approaches offline including FFT-based matching, but none of
them worked particularly well (if at all). There is some literature on
IR-RGB fusion from descriptors, but I wasn't able to get it to work with my
data. Really we need a good synced dataset to work with - and maybe then we
could just train a CNN or something (e.g. one backbone network for each
modality and then train on the loss between the outputs).

Harsh188, 2024-07-22:
Hi, has there been any progress on this? I'm trying to sync RGB-IR capture with a Boson and BlackFly S. How can I do this using Flirpy?

jveitchmichaelis, 2024-07-23:
In this case, you should use hardware triggering to sync both cameras. Then this becomes a mostly solved problem.