LJMUAstroecology/flirpy #59: Uknown FFF record subtype

Tnecniv-Teikram, 2021-09-15:
Hi, 

I got a folder with 4 .seq files that I'm trying to split using the split_seqs command.

The process starts with the first image but then stops at 3% and throws the following errors:
Anyone has a clue how to fix this?

  File "split_seqs", line 106, in <module>
    folders = splitter.process(files)
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/flirpy-0.2.3-py3.8.egg/flirpy/io/seq.py", line 131, in process
    self._process_seq(seq, folder)
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/flirpy-0.2.3-py3.8.egg/flirpy/io/seq.py", line 184, in _process_seq
    for frame in tqdm(Seq(input_file)):
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/tqdm/std.py", line 1185, in __iter__
    for obj in iterable:
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/flirpy-0.2.3-py3.8.egg/flirpy/io/seq.py", line 76, in __getitem__
    return Fff(chunk)
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/flirpy-0.2.3-py3.8.egg/flirpy/io/fff.py", line 28, in __init__
    self._find_data_offset()
  File "/home/vincentmarkiet/miniconda3/envs/fortum-heat-detection/lib/python3.8/site-packages/flirpy-0.2.3-py3.8.egg/flirpy/io/fff.py", line 126, in _find_data_offset
    raise IOError("Unknown FFF record subtype")
OSError: Unknown FFF record subtype

jveitchmichaelis, 2021-09-15:
Apologies about this - I'm going to add a fix for this (and other issues) by falling back to Exiftool if the built-in FFF parser doesn't work. For now, could you try an older version of flirpy?

Tnecniv-Teikram, 2021-09-15:
Thanks for the quick response. Will try that.

jveitchmichaelis, 2021-09-19:
@Tnecniv-Teikram can you try pulling the latest `master` and running split seqs again? This time, please provide the `--width` and `--height` parameters.

jveitchmichaelis, 2023-07-04:
Closing this as I believe this is now fixed in `main`; see the other thread. Please feel free to re-open if the problem's still there.

closed 2023-07-04 by jveitchmichaelis