LJMUAstroecology/flirpy #62: .seq from FLIR research Studio not processed

fclairec, 2021-11-30:
Hi, this is not an issue, but rather a question. 

I have a .seq file recorded by Flir Research Studio (https://syncandshare.lrz.de/getlink/fi36fXbGE9pyqEVuDSfqeDrh/Rec-FLIR_A50_89800406-308_14_33_55_913-0120-IR.seq). 

I have only little knowledge about the structure of the files, but it seems to be in a different format than what is required by this library. I would be willing to extend your library in a way that it is abe to read this file too. However I would appreciate some pointers on what might be the issue here. 
The video has 51 frames and the log after calling the splitter says: 

 ```
 0%|          | 0/1 [00:00<?, ?it/s]
  0%|          | 0/51 [00:00<?, ?it/s]
  0%|          | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "C:/Projekte/tir_exploration/exploration.py", line 10, in <module>
    sp.process(input_location)
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\flirpy\io\seq.py", line 144, in process
    self._process_seq(seq, folder)
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\flirpy\io\seq.py", line 197, in _process_seq
    for count, frame in enumerate(tqdm(Seq(input_file, self.height, self.width))):
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\tqdm\std.py", line 1180, in __iter__
    for obj in iterable:
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\flirpy\io\seq.py", line 79, in __getitem__
    return Fff(chunk, self.width, self.height)
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\flirpy\io\fff.py", line 41, in __init__
    self._get_image_size()
  File "C:\Users\ge37mof\Anaconda3\envs\tir_36\lib\site-packages\flirpy\io\fff.py", line 81, in _get_image_size
    assert res[1] == res[3]+1
AssertionError

Process finished with exit code 1

```

Thanks a lot in advance.

BR

jveitchmichaelis, 2023-07-04:
Hi @fclairec apologies for the very long delay,

I think this issue should now be fixed in `main`. If you still have that SEQ file lying around it'd be great if you could try this again to check that the images are correctly loaded.

juuiceboxx, 2024-06-20:
Hello, I am new to this project, but it would appear that I am having the same issue as **fclairec**. I am trying to split .seq recorded using a Flir A50 into single jpeg/png images for each frame with a matching csv file that includes the temp of each pixel value, and this tool is the best I have found so far using Python. I attempted in both Jupyter and the terminal, but both times I receive this assertion error. I also attempted both inputting my own height/width and allowing it to try to find its own, but both gave me the same error. 

Input: 
`python split_seqs.py -i Rec-0027.seq --split_filetypes --export_raw
`
```
INFO:__main__:Loading: Rec-0027.seq
INFO:flirpy.io.seq:Splitting 1 files

  0%|         | 0/1 [00:00<?, ?it/s]INFO:flirpy.io.seq:Splitting Rec-0027.seq into C:\Users\me\Ultralytics\seq_manage\Rec-0027
  0%|         | 0/921 [00:00<?, ?it/s]
  0%|         | 0/1 [00:00<?, ?it/s]

Traceback (most recent call last):
  File "C:\Users\me\AppData\Roaming\Python\Python311\Scripts\split_seqs.py", line 110, in <module>
    folders = splitter.process(files)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\flirpy\io\seq.py", line 144, in process
    self._process_seq(seq, folder)
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\flirpy\io\seq.py", line 197, in _process_seq
    for count, frame in enumerate(tqdm(Seq(input_file, self.height, self.width))):
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\tqdm\std.py", line 1181, in __iter__
    for obj in iterable:
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\flirpy\io\seq.py", line 79, in __getitem__
    return Fff(chunk, self.width, self.height)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\flirpy\io\fff.py", line 41, in __init__
    self._get_image_size()
  File "C:\Users\me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\flirpy\io\fff.py", line 81, in _get_image_size
    assert res[1] == res[3]+1
           ^^^^^^^^^^^^^^^^^^
AssertionError
```

jveitchmichaelis, 2024-06-21:
Hi, which version of flirpy are you using?

juuiceboxx, 2024-06-21:
Hello, thank you for the fast reply. It turns out that I was using an older version. Using "pip install flirpy" installs version 0.3.0, but using 0.4.0 fixed the issue.

jveitchmichaelis, 2024-06-22:
Great to hear, thanks for confirming. I'll push a new version to pypi!

closed 2024-06-25 by jveitchmichaelis