LJMUAstroecology/flirpy #63: Multiple Errors Loading SEQ File

leo-hstone, 2022-01-26:
Hi everybody,

I am new to python and thermal imaging. I hope this error is trivial an can be resolved easily.

I recorded a .seq file with a T420 FLIR Camera. Running `ython .\flirpy\scripts\split_seqs --width 320 --height 240` in the folder produced the following output: 
[error.log](https://github.com/LJMUAstroecology/flirpy/files/7945764/error.log)

Unfortunately I am not experienced enough to debug the error myself. It seems to be some problem with the meta data. After playing around with the cli arguments I managed to create the .txt files in the raw folder. All meta data listed there is correct.

Can you guide me in the right direction to fix this or is it an error with the general code? If needed I can supply the relevant files, but I would prefer to send them privately.

leo-hstone, 2022-01-26:
After playing a bit around with the python files, I can conclude, that the problem is probably the seq.py file.

```python
import numpy as np
from flirpy.io.seq import Seq
import matplotlib.pyplot as plt

#test = Seq("testrec1.seq", height=320, width=240) #as soon as supplied, error!
test = Seq("testrec1.seq")
print (len(test))

file0 = test[0]
file0.get_meta() # prints fine!

rad_image = file0.get_radiometric_image()
plt.imshow(rad_image) # looks great!

file1 = test[1] #test last file
print(file1.get_meta()) # prints same error!
```

Any help welcome!

jveitchmichaelis, 2022-01-26:
Any idea what the difference between your two frames are? e.g. test[0] works but test[1] doesn't?

leo-hstone, 2022-01-26:
As it is a video, I hope it is just the timestamp.

jveitchmichaelis, 2022-01-26:
A SEQ is just a series of frames encoded as FFF files - it's basically a stack of images in a single file. So there isn't really a difference between a single frame SEQ and a multi-frame SEQ.

I'd suggest rolling back to an earlier release of flirpy and see if that helps (e.g. 0.1 or 0.2). There might be API changes, but the split script in the repo should work. This is kind of a longstanding issue that I've not had time to fix - basically the current SEQ decoder seems to bug out with the endian-ness of certain files. It seems that SEQ files made by different cameras are not identical, which is annoying. Older versions of flirpy used Exiftool to do the splitting and that seemed more reliable.

leo-hstone, 2022-01-26:
I will check and try! Thank you very much for your tool and help :)

zipy124, 2022-01-28:
I also have errors when Loading a SEQ file from a A35 FLIR camera. The .seq is generated using the Research IR program. When not supplying a width and height, the meta data prints correctly except the width and height but calling get_radiometric_image() results in a blank array. When supplying correct width and height, the program errors out when trying to access the array. The error looks like the following:

```
<ipython-input-14-8d3be0f8dfd3> in <module>
----> 1 rad_image = data[50].get_radiometric_image()
      2 print(data[50].get_meta())
      3 plt.imshow(rad_image)

~/.conda/envs/tipa_env/lib/python3.8/site-packages/flirpy/io/seq.py in __getitem__(self, index)
     77         chunk = self.seq_blob[offset:offset+chunksize]
     78 
---> 79         return Fff(chunk, self.width, self.height)
     80 
     81 class Splitter:

~/.conda/envs/tipa_env/lib/python3.8/site-packages/flirpy/io/fff.py in __init__(self, data, width, height)
     32             self.height = height
     33             self.meta = {}
---> 34             self._find_data_offset_simple(width, height)
     35         else:
     36             try:

~/.conda/envs/tipa_env/lib/python3.8/site-packages/flirpy/io/fff.py in _find_data_offset_simple(self, width, height)
     89                     + struct.pack("<H", height-1)
     90 
---> 91         valid = re.compile(search)
     92         res = valid.search(self.data)
     93 

~/.conda/envs/tipa_env/lib/python3.8/re.py in compile(pattern, flags)
    250 def compile(pattern, flags=0):
    251     "Compile a regular expression pattern, returning a Pattern object."
--> 252     return _compile(pattern, flags)
    253 
    254 def purge():

~/.conda/envs/tipa_env/lib/python3.8/re.py in _compile(pattern, flags)
    302     if not sre_compile.isstring(pattern):
    303         raise TypeError("first argument must be string or compiled pattern")
--> 304     p = sre_compile.compile(pattern, flags)
    305     if not (flags & DEBUG):
    306         if len(_cache) >= _MAXCACHE:

~/.conda/envs/tipa_env/lib/python3.8/sre_compile.py in compile(p, flags)
    762     if isstring(p):
    763         pattern = p
--> 764         p = sre_parse.parse(p, flags)
    765     else:
    766         pattern = None

~/.conda/envs/tipa_env/lib/python3.8/sre_parse.py in parse(str, flags, state)
    946 
    947     try:
--> 948         p = _parse_sub(source, state, flags & SRE_FLAG_VERBOSE, 0)
    949     except Verbose:
    950         # the VERBOSE flag was switched on inside the pattern.  to be

~/.conda/envs/tipa_env/lib/python3.8/sre_parse.py in _parse_sub(source, state, verbose, nested)
    441     start = source.tell()
    442     while True:
--> 443         itemsappend(_parse(source, state, verbose, nested + 1,
    444                            not nested and not items))
    445         if not sourcematch("|"):

~/.conda/envs/tipa_env/lib/python3.8/sre_parse.py in _parse(source, state, verbose, nested, first)
    666                 item = None
    667             if not item or item[0][0] is AT:
--> 668                 raise source.error("nothing to repeat",
    669                                    source.tell() - here + len(this))
    670             if item[0][0] in _REPEATCODES:

error: nothing to repeat at position 0
```

zipy124, 2022-01-28:
I have now tried older versions of FlirPY and the error is exactly the same

jveitchmichaelis, 2022-02-14:
Interesting, perhaps this is something we could include as an option with flirpy? Do you know what the license terms are?

leo-hstone, 2022-04-19:
You can actually just download it here and use it (just create an account with a temp email): https://flir.custhelp.com/app/account/fl_download_software 

I did not have to enter any licenses.

jveitchmichaelis, 2022-04-19:
Thanks, I'll take a look - which package is it?

For example we could just provide an import and ask the user to install it in their python environment. Then I don't think we'd need to worry about licensing, we just call functions from the SDK.

leo-hstone, 2022-04-20:
This sounds like a good idea! The provided examples give a good introduction to the API package.