LJMUAstroecology/flirpy #64: NotADirectoryError: [WinError 267] The directory name is invalid

Hoogashaga, 2022-02-08:
Hi,
I still have below error due to NotADirectoryError: [WinError 267] 
May I know how to solve the problems? You can read the log I posted.

PS C:\XXXXX\venv> **python .\scripts\split_seqs.py --input "*.SEQ" --output "out" --width 640 --height 480**
INFO:__main__:Loading: test.seq
INFO:__main__:Loading: sample.seq
INFO:flirpy.io.seq:Splitting 2 files
  0%|                                                                                                                                                                                                              | 0/2 [00:00<?, ?it/s]I
NFO:flirpy.io.seq:Splitting test.seq into C:\XXXXX\venv\out\test
0it [00:00, ?it/s]
INFO:flirpy.io.seq:Extracting metadata
  0%|                                                                                                                                                                                                              | 0/2 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "C:\XXXXX\venv\scripts\split_seqs.py", line 112, in <module>
    **folders = splitter.process(files)**
  File "C:\XXXXX\venv\lib\site-packages\flirpy\io\seq.py", line 163, in process
    self.exiftool.write_meta(filemask)
  File "C:\XXXXX\venv\lib\site-packages\flirpy\util\exiftool.py", line 75, in write_meta
    res = subprocess.call(cmd, cwd=cwd, stderr=subprocess.PIPE, stdout=subprocess.PIPE)
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
**NotADirectoryError: [WinError 267] The directory name is invalid**

jveitchmichaelis, 2022-02-08:
Can you try with an absolute path? And make sure the folder exists?

Hoogashaga, 2022-02-08:
How to know my current version?　I installed the lib a week ago by "pip install flirpy". I think it's the latest version.
Yeah, I tried like "python C:\XXXXX\venv\Scripts\split_seqs.py --input "*.SEQ" --output "out" --width 640 --height 480"
But same problem, **NotADirectoryError: [WinError 267] The directory name is invalid**

jveitchmichaelis, 2022-02-08:
I meant for the output folder, e.g. "C:\XXXXX\out" and check that out exists. Yeah if it's from pip it's probably OK. I believe that error means something is being passed a filename instead of a directory. Could you maybe edit the file (C:\XXXXX\venv\lib\site-packages\flirpy\util\exiftool.py) and put a print statement where you get the error (line 75) to see what `cmd` and `cwd` are?

Hoogashaga, 2022-02-08:
The out is generated with subfolder having same name as my SEQ file, but it's empty in the subfolder
cmd=>**['C:\\XXXXX\\venv\\lib\\site-packages\\flirpy\\bin\\exiftool.exe', '-ext', 'fff', '.', '-w!', '.txt']**
cwd=>**C:\XXXXX\out\sample\raw**

jveitchmichaelis, 2022-02-08:
And can you try running that command manually?

Hoogashaga, 2022-02-08:
"_can you try running that command manually_" =>I'm not sure what you mean here
Yes, raw folder is empty. Same problem.  
I also print line 112 in split_seqs.py
=> **['C:\\XXXXX\\venv\\sample.SEQ']**

jveitchmichaelis, 2022-02-08:
Underneath, all flirpy is doing is running a chain of Exiftool commands on your data. Actually the version you have may be using flirpy's SEQ decoder which is possibly failing, though I don't think that's the problem here. You'd be getting an error later on if that were the case. It seems like right now it's failing to split a single file.

So, first we take each SEQ file and we split into FFF files - this step seems to have failed, not sure why.

Then flirpy will use Exiftool to extract metadata from each of the fff files (i.e. fff -> txt) and this is used for radiometry.

Finally, firpy will open each fff file in turn, extract the raw data and use the radiometry to save it as a png/tiff.

Hoogashaga, 2022-02-08:
Yes, it's 640*480
I tried below from exiftool forum 
_exiftool "C:\XXXXX\venv\Scripts\sample.seq" -b -RawThermalImage  >C:\output_
**Access is denied.**

Is it possible that  my company computer blocked any new write and then leading to NotADirectoryError: [WinError 267] The directory name is invalid ?

jveitchmichaelis, 2022-02-11:
It's possible, but permission errors can be fairly common with Python on Windows. Have you tried running powershell/prompt administrator?

Hoogashaga, 2022-02-11:
Yes, I used prompt administrator.

jveitchmichaelis, 2022-02-11:
If you're unable to modify the directory yourself (i.e. not using Python) then I'm not sure I can help until you figure out why that's happening, it doesn't seem to be an issue with flirpy specifically.