LJMUAstroecology/flirpy #67: ModuleNotFoundError: No module named 'flirpy.camera'

MrvKrt, 2022-03-22:
Even I've installed Flirpy with all requirements, I still receive this error code. Specifically, run:

from flirpy.camera.boson import Boson

camera = Boson()
image = camera.grab()
camera.close() 

Any help?

jveitchmichaelis, 2022-03-23:
Can you describe how you installed flirpy? Does it work within a conda/virtualenv?

MrvKrt, 2022-04-13:
I use `pip install flirpy` command as usual in Visual Studio Code 15. I also download the repository and use `pip install` command. But still receive the same error code.

MrvKrt, 2022-04-22:
Thank you for your interest. But even if I try on the COLAB, I receive an error again.

Here: https://colab.research.google.com/drive/1nVFP2cVk5_3dF8baD_K5KDll_ByNNsLC#scrollTo=cEfmPgeQtg9U

[image]

jveitchmichaelis, 2022-04-22:
Ok, I think it works now - you may need to fiddle with OpenCV and uninstall/install. I added a comment in your notebook

MrvKrt, 2022-04-24:
> 
I'm suspecting that this is possiblly because of the environmental issue, too. I tried the code for creating the new enviromental interpreter both VS and Spider and didn't work. Interestingly these codes work in Jupyter Notebook. 
```
camera = Boson()
image = camera.grab()
camera.close()
```
I didn't get the chance to try it with my thermal camera. Probably because of that I've received an error "Boson can't open". I'll check it tomorrow with my thermal, but I think this time it's gonna work. :)