LJMUAstroecology/flirpy #69: Flir P640

glebpinigin, 2022-05-21:
Is there any straightforward way to open .seq files written by Flir P640?

I have some old .seq files without any information of camera (only assumption that this camera is P640)

flirpy fails to read camera and lens metadata, however successfully reads radiometric metadata and returns several shifted arrays of pixels before crash. Isn't there any straightforward way to hardcode P640's metadata or do anything else, that could possibly help?

closed 2022-05-30 by glebpinigin

jveitchmichaelis, 2022-05-30:
Hiya the short answer is try an older release of flirpy, or try to extract the file directly with Exiftool?

glebpinigin, 2022-06-10:
Hi! I have tried another solution provided by https://github.com/gtatters/ThermimageBash and it succeeded. There is a perl script that batch process sequence in correct way. As I can guess, this script is calling Exiftool directly. I have tried to open fff files, which were correctly created py this perl script with flirpy and and it resulted in the same issue. They also have an R [package](https://github.com/gtatters/Thermimage). But in case of input/output it is a wrapper over Exiftool, as I understood from code and docs.

Hope my answer is helpful!

jveitchmichaelis, 2022-06-10:
Yeah not too surprising. The original version of flirpy also just calls
exiftool. The main reason I changed it was that it doesn't offer an easy
way to iterate directly over fff files (maybe it does if you call it as a
library?). However there is some subtlety in the decoding it seems (due to
endianness), so it works for some files and not others.

The simplest approach at this point (as I've mentioned in other issues)
would be to roll back flirpy to allow a fallback. But unfortunately I just
don't have the time to do this right now.

glebpinigin, 2022-06-10:
If I understood everything correctly, iteration over fff files, that are part of sequense, can be solved by saving them as temporary files and iterating over files in file system. I am not sure that it is a good way to process .seq file as a video, but in general that might work. Another thing is that P640 is a completely outdated camera and perhaps not worth it.

jveitchmichaelis, 2022-06-10:
Yes and this is exactly how flirpy solved it - split a sequence file into
separate FFF files and then deal with them iteratively (since exiftool can
also convert to other formats and extract metadata) :)

I don't think it's necessarily a difficult problem, it just requires some
time to write a good header decoder (eg as exiftool has). I did try to
design flirpy's based on exiftool's code. It's also tricky because the
format doesn't seem to be super standard so it requires a lot of file
samples to get a universal solution.