LJMUAstroecology/flirpy #70: AttributeError when trying to do a simple grab on Mac OS

carloscastellanos, 2022-06-29:
Trying to run the simple grab code gives me the following error:

```
Traceback (most recent call last):
  File "capture_test.py", line 24, in <module>
    image = camera.grab()
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 156, in grab
    self.setup_video(device_id)
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 94, in setup_video
    device_id = self.find_video_device()
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 46, in find_video_device
    if device.contains("VendorID_1E4E") and device.contains("ProductID_0100"):
AttributeError: 'str' object has no attribute 'contains'
```

Lepton 3.5
PureThermal-2 board
Mac OS 10.15.6
Python 3.7.13

jveitchmichaelis, 2022-06-29:
Should be fixed https://github.com/LJMUAstroecology/flirpy/commit/b278ad4e0ba91efb621e16bf68f7212b5cc309aa

jveitchmichaelis, 2022-06-29:
> Thanks, but now it says the camera is not connected.
>
> Traceback (most recent call last):
>   File "capture_test.py", line 24, in <module>
>     image = camera.grab()
>   File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 157, in grab
>     self.setup_video(device_id)
>   File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 98, in setup_video
>     raise ValueError("Lepton not connected.")
> ValueError: Lepton not connected.

Can you print the connected device IDs? Either in the code or with lsusb or
similar

jveitchmichaelis, 2022-06-29:
Actually you can just run the command the script does (system_profiler SPCameraDataType)

jveitchmichaelis, 2022-06-29:
This is a bit odd. The output from system profiler before looks fine:

```
Product ID: 0x0100
Vendor ID: 0x1e4e  (Etron Technology, Inc.)
```

But here you have a different PID/VID:

```
UVC Camera VendorID_7758 ProductID_256
```

I guess we ought to check for both, but I'm not sure why it's enumerating as different things.

carloscastellanos, 2022-06-29:
Well 7758 is the hex to decimal translation of 0x1e4e. I guess Apple converts it for some reason?

Anyways, I made the changes you suggested but it still didn't work. I get the following errors now:

using camera.setup_video(1):

```
[ERROR:0] global /private/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/pip-req-build-tetsazc6/opencv/modules/videoio/src/cap.cpp (291) open VIDEOIO(AVFOUNDATION): raised unknown C++ exception!

Traceback (most recent call last):
  File "capture_test.py", line 25, in <module>
    camera.setup_video(1)
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 113, in setup_video
    raise IOError("Failed to open capture device {}".format(device_id))
OSError: Failed to open capture device 1
```

with only the changes to lepton.py:

```
Traceback (most recent call last):
  File "capture_test.py", line 26, in <module>
    image = camera.grab()
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 166, in grab
    self.decode_telemetry(image, telemetry_mode)
  File "/Users/ccastellanos/opt/anaconda3/envs/artibmcp-thermal/lib/python3.7/site-packages/flirpy/camera/lepton.py", line 123, in decode_telemetry
    res = struct.unpack("<2cII16x4h6xIh2xh8xhI4xhhhhhh64xI172x", image[-2,:])
struct.error: unpack requires a buffer of 320 bytes
```
and the builtin-in webcam turns on briefly

jveitchmichaelis, 2022-06-29:
Odd well I guess on the macbook I initially used for this it didn't. I don't own a PT2 so I can't verify unfortunately.

If your webcam is blinking then you're probably using the wrong device ID. Did you try 0?

carloscastellanos, 2022-06-29:
Yes I tried 0 as well. I'm beginning to think this may be an issue with my Macbook. I'm going to try this on another Mac and see what happens

jveitchmichaelis, 2022-06-29:
Can you try with OpenCV directly?

Flirpy just creates a VideoCapture object. The GroupGets library (I think?) uses, or at least used to use, a custom UVC driver so they got around limitations with Mediakit.

carloscastellanos, 2022-06-29:
Yes I tried OpenCV as well and that also is not working! I'm getting "an unknown C++ exception". 

I've tried installing the groupgets version of libuvc (thinking that might be the issue) but that fails too when I try to run cmake.  So not a good for me, lol.

That's actually my original plan. Capture with OpenCV and use flirpy for analysis.

I've also posted on the groupgets googlegroup, so I'm still hopeful I can this working.

carloscastellanos, 2022-07-05:
UPDATE: I think I may have figured it out. If run as sudo, this file works: https://github.com/groupgets/purethermal1-uvc-capture/blob/master/python/uvc-radiometry.py

This grabs a stream directly from libuvc. So it seems that at least for now libuvc is the way to go on pure-thermal 2 on MacOS for grabbing a video stream

jveitchmichaelis, 2022-07-05:
Yeah it's possible we could bundle libuvc with flirpy and have it load on Mac somehow. I'll see if I can get hold of a PT2 board to test on my laptop, but for now I'd suggest just using the groupgets library.

jveitchmichaelis, 2022-10-03:
Closing as this seems to be a somewhat-isolated issue, until we can definitively fix it here there is a solution in the other library.

jveitchmichaelis, 2022-10-11:
> I'm on a Mac so no lsusb so I used system_profiler and it's there:
>
> USB:
>
>     USB 3.1 Bus:
>
>       Host Controller Driver: AppleIntelCNLUSBXHCI
>       PCI Device ID: 0xa36d
>       PCI Revision ID: 0x0010
>       PCI Vendor ID: 0x8086
>
>         USB2.0 HUB:
>
>           Product ID: 0x0101
>           Vendor ID: 0x1a40  (TERMINUS TECHNOLOGY INC.)
>           Version: 1.00
>           Speed: Up to 480 Mb/s
>           Location ID: 0x14200000 / 1
>           Current Available (mA): 500
>           Current Required (mA): 100
>           Extra Operating Current (mA): 0
>
>             USB Storage:
>
>               Product ID: 0x0751
> [... 68 lines cut ...]
>       |       "USB Serial Number" = "00190014-5119-3038-3532-373600000000"
>       |       "kUSBVendorString" = "GroupGets"
>       |       "IOClassNameOverride" = "IOUSBDevice"
>       |     }
>       |

OK then I suggest printing out each device in the loop to see what's being
found (and what the code is testing against)