LJMUAstroecology/flirpy #73: split_seqs command returns conflicting option string error

mattxcnm, 2022-09-13:
```
(env) C:\Users\[project repository]>python .\env\scripts\split_seqs -h 
Traceback (most recent call last):
  File "C:\Users\[project repository]\env\scripts\split_seqs", line 70, in <module>
    parser.add_argument('-h', '--height', help='Image height', type=int, default=None)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1441, in add_argument
    return self._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1807, in _add_action
    self._optionals._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1643, in _add_action
    action = super(_ArgumentGroup, self)._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1455, in _add_action
    self._check_conflict(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1592, in _check_conflict
    conflict_handler(action, confl_optionals)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2032.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1601, in _handle_conflict_error
    raise ArgumentError(action, message % conflict_string)
argparse.ArgumentError: argument -h/--height: conflicting option string: -h
```

jveitchmichaelis, 2022-09-13:
Hi, please could you provide more info about what you're trying to do?

mattxcnm, 2022-09-14:
@jveitchmichaelis I followed the directions for pip installation of the library with pip v22.2.2 and a Python v3.10.7 virtual environment.

I ran the command `python .\env\scripts\split_seqs -h` from a command prompt terminal in my Python virtual environment to try and get the splitter command to display the help menu, but received the error shown above.

My goal is to use flirpy to make a 2.75TB seq file generated by a FLIR A50 to be more manageable in a coding environment.

I will try another Python version. Thanks!

mattxcnm, 2022-09-14:
### Python 3.7.9 & PIP 20.1.1

ran `python .\.env37\scripts\split_seqs -h` in virtual environment
```
(.env37) [Repository Path]>python .\.env37\scripts\split_seqs -h 
Traceback (most recent call last):
  File ".\.env37\scripts\split_seqs", line 3, in <module>
    import flirpy.io.seq
  File "[Repository Path]\.env37\lib\site-packages\flirpy\io\seq.py", line 18, in <module>
    from flirpy.io.fff import Fff
  File "[Repository Path]\.env37\lib\site-packages\flirpy\io\fff.py", line 5, in <module>
    import six
ModuleNotFoundError: No module named 'six'
```
used `pip3 install six`

ran `python .\.env37\scripts\split_seqs -h` in virtual environment

```
(.env37) [Repository Path]>python .\.env37\scripts\split_seqs -h
Traceback (most recent call last):
  File ".\.env37\scripts\split_seqs", line 70, in <module>
    parser.add_argument('-h', '--height', help='Image height', type=int, default=None)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1373, in add_argument
    return self._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1736, in _add_action
    self._optionals._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1577, in _add_action
    action = super(_ArgumentGroup, self)._add_action(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1387, in _add_action
    self._check_conflict(action)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1526, in _check_conflict
    conflict_handler(action, confl_optionals)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\argparse.py", line 1535, in _handle_conflict_error
    raise ArgumentError(action, message % conflict_string)
argparse.ArgumentError: argument -h/--height: conflicting option string: -h
```
I have Python 3.10, 3.9, 3.8, and 3.7 for other testing (All versions downloaded from Windows store, not directly from Python.org)

jveitchmichaelis, 2022-09-14:
OK can you just remove the height/width lines from the script for now? The latest interface shouldn't require knowledge of image width/height (it's inferred from the image).