LJMUAstroecology/flirpy #75: ValueError: Lepton not connected.

EduardoRonchi, 2022-09-30:
Hello. I am using a PT Mini + Lepton 3.5 with a Raspberry Pi 4b and I am constantly getting the following error:
`  File "/usr/local/lib/python3.9/dist-packages/flirpy/camera/lepton.py", line 156, in grab
    self.setup_video(device_id)
  File "/usr/local/lib/python3.9/dist-packages/flirpy/camera/lepton.py", line 97, in setup_video
    raise ValueError("Lepton not connected.")
ValueError: Lepton not connected.`

I don't believe it is a misconnection since I have 2 setups with the same error. 
What may I do?

Thanks in advance,
Eduardo.

EduardoRonchi, 2023-03-14:
Hello, the device is seen by Linux. The Lepton works when I reboot the OS, but after some time it throws the error above.

EduardoRonchi, 2023-03-14:
Mistankenlly closed the issue.

jveitchmichaelis, 2023-03-14:
Hi Eduardo.

This is quite difficult to debug without access to the hardware, but we can try. If you look at the code for flirpy, all we're doing is opening the camera with OpenCV.

Next time the camera crashes, can you try to load it separately before you reboot? Eg using the VideoCapture module in opencv. Similarly you could try a very simple loop where you read from the camera repeatedly until it crashes. This would hopefully isolate the bug to opencv or flirpy.

Is the device seen by Linux after the crash?

It's possible that somehow flirpy is reading a value that's putting the camera into a weird state, but we don't send any commands to the Lepton - we just request video. So I think that's unlikely as it's not the same as a Boson where you can set/change the camera config.

Best

EduardoRonchi, 2023-03-14:
Hello,

When the error occur I get the following warning:
`[ WARN:0@10.531] global /io/opencv/modules/videoio/src/cap_[v4l.cpp](https://v4l.cpp/) (1000) tryIoctl VIDEOIO(V4L2:/dev/video1): select() timeout.`

Then I tried to get the images from the camera using OpenCV with 
```
cap = cv2.VideoCapture(1)
ret, frame = cap.read()
```
But I got the sabe warning as above.

The camera is "attatched" to usb, issueing the command v4l2-ctl --list-devices i get the following:

[image]

At /dev/video0 I have the CSI-camera from raspberry py and at /dev/video1 and 2 I have the purethremal camera.

I am afraid the problem is not in flirpy library, but you have any idea to solve this?

Best regards,
Eduardo

jveitchmichaelis, 2023-03-14:
I'd suggest raising an issue with Pure Thermal or OpenCV - and also try it out on a different platform like a laptop to see if it's just an issue on the Pi