LJMUAstroecology/flirpy #80: Merging folders phase is failing

Tnecniv-Teikram, 2023-01-09:
Hi,

I tried to extract a .seq file but after the frame extraction phase the merging folder phase fails. I'm not sure what exactly is the main error here. The program has write permission and the .raw folder has .txt for each frame.

Anyone got a clue what is going wrong?

Thanks,
Vincent

(flirpy) ➜  02_01_2023 git:(master) ✗ python3 ~/flirpy/scripts/split_seqs -i seq1.seq -o .
INFO:__main__:Loading: seq1.seq
INFO:flirpy.io.seq:Splitting 1 files
  0%|                                                                                 | 0/1 [00:00<?, ?it/s]INFO:flirpy.io.seq:Splitting seq1.seq into /home/vincentmarkiet/flirpy/data/fortum/Espoo_jan_2023/seq_files/02_01_2023/seq1
                                                                                                           WARNING:root:Failed to extract camera information                                      | 0/3 [00:00<?, ?it/s]
--- Logging error ---
Traceback (most recent call last):
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 245, in get_meta
    meta["Camera Model"] = res[0].decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8c in position 8: invalid start byte

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
    msg = self.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
    return fmt.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 112, in <module>
    folders = splitter.process(files)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 144, in process
    self._process_seq(seq, folder)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 197, in _process_seq
    for count, frame in enumerate(tqdm(Seq(input_file, self.height, self.width))):
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/tqdm/std.py", line 1180, in __iter__
    for obj in iterable:
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 79, in __getitem__
    return Fff(chunk, self.width, self.height)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 44, in __init__
    self.meta = self.get_meta()
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 251, in get_meta
    logger.warn("String: ", res)
Message: 'String: '
Arguments: ((b'C&G&]&v&\x8c&\xae&\xe2&\xeb&\xde&\xe6&\xdb&\xe9&\xea&\xe2&\xe0&\xfa&', b'\xd9&\xd9&\xcf&\xa8&\xa7&\x97&\x90&\x96&', b'\x97&|&t&w&b&c&_&a&', b'X&^&R&}&z&~&\x8d&\x92&', b"\x92&\x92&\x9d&\xaf&\xa8&\xae&\xb6&\xd7&\xef&\xf9&\xfe&\xfa&\t'\x02'\x13'\x0b'", b"\x0e'\xf9&\xf8&\xec&\xee&\xde&\xcb&\xab&", b'x&p&J&!&\x18&3&9&2&'),)
WARNING:root:Failed to extract lens details
--- Logging error ---
Traceback (most recent call last):
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 256, in get_meta
    meta["Lens Model"] = res[0].decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x96 in position 0: invalid start byte

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
    msg = self.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
    return fmt.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 112, in <module>
[... 80 lines cut ...]
    for obj in iterable:
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 79, in __getitem__
    return Fff(chunk, self.width, self.height)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 44, in __init__
    self.meta = self.get_meta()
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 262, in get_meta
    logger.warn("String", res)
Message: 'String'
Arguments: ((b'\x9a&\xaa&\x9b&\xa1&\x97&\x98&\x7f&t&|&w&|&\x84&\x8b&\x8a&\x8d&\xa4&', b'\x90&\xa3&\xb1&\xb5&\xb3&\xae&\xb7&\xa8&', b"\x97&\xbe&\xc6&\xdf&\xfa&\x04'\x13'\xfd&", 2.0837866005732515e-15),)
100%|█████████████████████████████████████████████████████████████████████████| 3/3 [00:00<00:00, 42.32it/s]
INFO:flirpy.io.seq:Extracting metadata
INFO:flirpy.io.seq:Copying tags to radiometric
INFO:flirpy.io.seq:Copying tags to preview
100%|█████████████████████████████████████████████████████████████████████████| 1/1 [00:01<00:00,  1.01s/it]
INFO:__main__:Merging folders
  0%|                                                                                 | 0/1 [00:00<?, ?it/s]INFO:__main__:Removing raw folder from /home/vincentmarkiet/flirpy/data/fortum/Espoo_jan_2023/seq_files/02_01_2023/seq1
  0%|                                                                                 | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 121, in <module>
    shutil.copy2(os.path.join(folder, "raw", "*.txt"), os.path.join(folder, "radiometric"))
  File "/usr/lib/python3.8/shutil.py", line 435, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/vincentmarkiet/flirpy/data/fortum/Espoo_jan_2023/seq_files/02_01_2023/seq1/raw/*.txt'

Tnecniv-Teikram, 2023-01-09:
Partly solved this problem by adding the option --export_raw

Tnecniv-Teikram, 2023-01-09:
Each frame is black except for the first image for each .seq file. 

Flirpy seems to fail extracting camera information. 

```
--- Logging error ---
Traceback (most recent call last):
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 257, in get_meta
    meta["Lens Part Number"] = res[1].decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfe in position 0: invalid start byte

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
    msg = self.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
    return fmt.format(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 112, in <module>
    folders = splitter.process(files)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 144, in process
    self._process_seq(seq, folder)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 197, in _process_seq
    for count, frame in enumerate(tqdm(Seq(input_file, self.height, self.width))):
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/tqdm/std.py", line 1180, in __iter__
    for obj in iterable:
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/seq.py", line 79, in __getitem__
    return Fff(chunk, self.width, self.height)
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 44, in __init__
    self.meta = self.get_meta()
  File "/home/vincentmarkiet/.local/lib/python3.8/site-packages/flirpy/io/fff.py", line 262, in get_meta
    logger.warn("String", res)
Message: 'String'
Arguments: ((b'\x0f&\x0f&\'&"&\x17&\x15&\x10&\x12&\x08&\x15&\x05&\x12&\t&\x12&\x0c&\x07&', b'\xfe%\n&\xf8%\xf2%\xf6%\xf2%\xdf%\xe4%', b'\xe7%\xdd%\xe2%\xdc%\xde%\xd6%\xe5%\xdc%', 4.3567229639275584e-16),)
100%|█████████████████████████████████████████████████████████████████████| 217/217 [00:04<00:00, 52.75it/s]
INFO:flirpy.io.seq:Extracting metadata
INFO:flirpy.io.seq:Copying tags to radiometric
INFO:flirpy.io.seq:Copying tags to preview
100%|█████████████████████████████████████████████████████████████████████████| 1/1 [00:13<00:00, 13.54s/it]
INFO:__main__:Merging folders
  0%|                                                                                 | 0/1 [00:00<?, ?it/s]INFO:__main__:Copying /home/vincentmarkiet/flirpy/data/fortum/Espoo_jan_2023/seq_files/test
100%|█████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  2.48it/s]
INFO:__main__:217 infrared frames
0it [00:00, ?it/s]
Traceback (most recent call last):
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 160, in <module>
    assert get_frame_number(n_rgb) == n_infrared
  File "/home/vincentmarkiet/flirpy/scripts/split_seqs", line 158, in <lambda>
    get_frame_number = lambda i: round(n_infrared*float(i)/n_rgb)
ZeroDivisionError: float division by zero
```

Tnecniv-Teikram, 2023-01-10:
Hi, 

I uploaded a small .seq file with 3 frames. It's from a Flir A655sc. I created a conda environment and pip installed flirpy.

You can access the .seq file here: 
https://www.dropbox.com/scl/fo/7sxeaczvyu8phtbx0mbr5/h?dl=0&rlkey=qkakft7iv1rr1kh8w5an06qwk

Thanks in advance.

Tnecniv-Teikram, 2023-02-19:
Any chance you've been able to take a look at it?

a4452, 2023-07-04:
I have the same (or very similar) issue. All my frames are black when using the split script.

Here is the beginning of the output:
[out-head.txt](https://github.com/LJMUAstroecology/flirpy/files/11945996/out-head.txt)
[Test03_seq.zip](https://github.com/LJMUAstroecology/flirpy/files/11946006/Test03_seq.zip)

Could you, please, help me overcoming the issue?

jveitchmichaelis, 2023-07-04:
I would suggest trying an older version of Flirpy for now - also double check if you can retrieve information from the file using EXIFtool. If Flirpy is failing to extract camera calibration info then that's likely to be the problem.

jveitchmichaelis, 2023-07-04:
@a4452 I had a play around, but there is still some odd offset in the extracted image and I'm not sure where it's coming from:

[image]

jveitchmichaelis, 2023-07-04:
@Tnecniv-Teikram 

Does this look right?

[image]

a4452, 2023-07-04:
> @a4452 I had a play around, but there is still some odd offset in the extracted image and I'm not sure where it's coming from. It looks like your image has some additional header that we're not taking into account and perhaps we're not extracting the image region correctly.
> 
> [image]
> 
> Raw frame and metadata from Exiftool attached (rename the image to .fff)
> 
> [frame.zip](https://github.com/LJMUAstroecology/flirpy/files/11948020/frame.zip)
> 
> This is with Exiftool's extractor, so we might just use that as a fallback (`exiftool.exe -rawthermalimage -b *.fff -w .tiff`):
> 
> [image]
> 
> This is with Flirpy's extracted FFF file, so it seems that the extracted files are "correct". It just looks like the image offset we're computing from within that FFF file is wrong (i.e. we're somehow including bits of the header in the image).

Thank you for your fast answer!
No, this is not correct. The first frame should look like (Thermal Studio):
[image]

It seems that the header is read correctly, at least the parameters you included in the text file (compare it to the right side of my printscreen). May, the pattern is similar to the original one, but with incorrect values ...

However, I'm not even able to get the frames you are showing (all my frames are black as the first one). In the raw directory, see the converted image
`exiftool -rawthermalimage -b frame_000000.fff -w frame_000000.tiff`
[Test03.zip](https://github.com/LJMUAstroecology/flirpy/files/11948925/Test03.zip)

jveitchmichaelis, 2023-07-04:
The values are not radiometric in the image that I showed, or it's off by a scale factor.

a4452, 2023-07-04:
> The values are not radiometric in the image that I showed - they're the raw counts on the sensor (or they are radiometric but they're scaled to integer values). But the pattern looks good, which is the important thing. I'm working on updating the code, might take a day or two for testing.

Thank you very much! No pressure, I don't need it asap :-). If I can provide more details or anything else, let me know.

jveitchmichaelis, 2023-07-04:
Ok made some progress. Basic decoding seems to work with both the supplied images, e.g. @Tnecniv-Teikram yours is also good.

[image]

a4452, 2023-07-07:
> 

Hi @jveitchmichaelis 

Perfect, I do get the same result as you posted. Also, other frames and files look ok. Thanks a lot!

Since you are the expert, I'll need to do some postprocessing (in python). What would be the best way to get numeric values, e.g., to a dataframe? Would there be an efficient way to get it from the split_seqs script (e.g., to export one text file for each script), or shall I do on the level of final plots?

jveitchmichaelis, 2023-07-07:
Depends what you need, you can call the Seq object yourself and you should be able to retrieve frames by index which will give you an Fff object that you can query to get the radiometric data.

jveitchmichaelis, 2023-07-08:
Closing as the main issue seems to be resolved, but feel free to respond if you have more questions

closed 2023-07-08 by jveitchmichaelis