LJMUAstroecology/flirpy #83: Inconsistent Output with FLIR Lepton 3.5

fishymonarch, 2023-05-07:
I'm trying to read temperatures from a FLIR Lepton 3.5 and I am having inconsistent readings. Here's the code:
```
from flirpy.camera.lepton import Lepton
cam = Lepton()
image = cam.grab()
print(image)
cam.close()
```

About half the time, the outputs looks as I'd expect:
```
[[29412 29444 29418 ... 29442 29414 29412]
 [29440 29440 29435 ... 29442 29420 29414]
 [29397 29427 29420 ... 29461 29414 29410]
 ...
 [29947 30004 30000 ... 29526 29502 29511]
 [29998 30024 30042 ... 29500 29511 29517]
 [30010 30036 30040 ... 29507 29491 29513]]
```

But the other half of the time it looks like this:
```
[[  128 13008   527 ...     0     0     0]
 [    0     0     0 ...     0  2048     0]
 [    0     0     0 ...     0 55733    14]
 ...
 [    0     0     0 ...     0     0     0]
 [    0     0     0 ...     0     0     0]
 [    0     0     0 ...     0     0     0]]
```

jveitchmichaelis, 2023-05-10:
Hi,

Normally in this situation I'd suggest reading with OpenCV directly and seeing if you can replicate. Flirpy is a thin wrapper so you can check that to see it it's a connectivity problem. Also have you checked with the GroupGets software? That might also rule out an issue with your camera 

Thanks!

jveitchmichaelis, 2023-09-01:
Closing as a stale issue, if you have any follow-up questions feel free to re-open

closed 2023-09-01 by jveitchmichaelis