LJMUAstroecology/flirpy #84: Frame Rate on Raspberry Pi

c-keil, 2023-05-18:
Is there anything in the flirpy code that would limit the frame rate to 30hz when using a raspberry pi with a FLIR Boson? I use the following script to assess frame rate, and it hovers right around 30 hz on a raspberry pi, but hits 60hz with the same camera on my laptop. 
```from flirpy.camera.boson import Boson
import cv2
import numpy as np
import time

loop_start_time = None
with Boson() as camera:
    while True:
        img = camera.grab().astype(np.float32)
        if loop_start_time is None:
            loop_start_time = time.time()
        else:
            current_time = time.time()
            rate = 1/(current_time-loop_start_time)
            loop_start_time = current_time
            print(rate)
```
As far as I can tell, none of the pi's cores are maxed out, and 30hz seems like a suspiciously round number. I took a quick look at the Boson code and didn't see anything obvious.

jveitchmichaelis, 2023-05-18:
Shouldn't be, the code was tested with a 60Hz Boson without any trouble. You could try querying the underlying OpenCV device to see what the frame rate is set to (or what it thinks it is).

c-keil, 2023-05-23:
When I check the OpenCV cap with something like:
```
from flirpy.camera.boson import Boson
import cv2
camera = Boson()
image = camera.grab()
print(camera.cap.getBackendName())
print(camera.cap.get(cv2.CAP_PROP_FPS))
```
I get:
```
V4L2
30.0
```
I get the same response on the Raspberry Pi and my laptop, but am able to grab frames at 60Hz on the laptop, and only at 30Hz on the Pi. So, seems like there is some underlying difference in OpenCV or the backend but it's not obvious what. I'll have to investigate further.

jveitchmichaelis, 2023-05-23:
Have you tried just setting the capture property to 60? You can check with
v4l2 on the Pi to see what formats are available?

`v4l2-ctl --list-formats`