LJMUAstroecology/flirpy #86: Radiometric data on Lepton 3.5

Hrishikesh-Kalyanaraman, 2023-08-01:
I am trying to get radiometric data from my lepton. I have successfully connected to my device and received a numpy array. However, I'm unsure what exactly this array means. It does not seem like it represents temperature data in micro kelvin like this issue mentions: https://github.com/LJMUAstroecology/flirpy/issues/53 as when I try to get the largest value from my array, it is always 58744 regardless of other data.

How do i get radiometric data from the lepton?

jveitchmichaelis, 2023-08-01:
Hi,

Please provide a sample image. I would guess 58744 is some telemetry value but it's impossible to say without seeing any data. If you have TLinear mode on, the results should be in Kelvin / 100.

Thanks!

Hrishikesh-Kalyanaraman, 2023-08-01:
How would I turn tlinear mode on? Is it automatically turned on with flirpy?

jveitchmichaelis, 2023-08-01:
If you're using the Pure Thermal board then you should use their configuration tools. Flirpy doesn't offer any config for the Lepton at the moment.

You could save it with `np.save` for example?

Hrishikesh-Kalyanaraman, 2023-08-02:
It seems like you were right. I checked data by row, and the last 2 rows are radiometric data. Thank you!

On the other hand, I seem to have an issue on Linux where the camera disconnects after I access it 10-12 times on Linux. I did not have this issue on Windows. I suspect it might be because Linux reads multiple video devices for some reason.

jveitchmichaelis, 2023-08-02:
Unfortunately my answer would be the same - speak to PureThermal or OpenCV because the opening/closing of the camera is done via those libraries, not Flirpy's. It's possible that we're not closing/releasing the camera properly so what you could try is writing your own code where you open and close the camera repeatedly and see if you can trigger this fault.

Flirpy should call `camera.release()` when you close the context (e.g. if using `with Lepton() as camera` or delete the object.