LJMUAstroecology/flirpy #87: boson radiometric

bakas90, 2023-08-04:
I'm using a Boson R camera.
I am getting raw data using flirpy.

However, I'm not exactly sure how to get the metadata.

I want to get the radiometric value.
I also want to create a SEQ file.

I would be grateful if you could tell me how.

jveitchmichaelis, 2023-08-04:
Hi, 

> I want to get the radiometric value.

Flirpy was written before the R variant was released and unfortunately I don't have a radiometric camera to test with.

Do you know what format the raw data from the camera is returned as? It should be straightforward to update the library to include radiometric support. Documentation is [here](https://flir.netx.net/file/asset/46046/original/attachment)

I'd guess we need to check if TLinear mode is enabled at least, and there seem to be options to control what units are output.

> I also want to create a SEQ file.

SEQ files are a sequence of FFF files. In theory you could construct these manually, but this isn't something that Flirpy supports - it's mostly used in the opposite direction (getting data out of SEQ files).

Is there a reason you want an SEQ file?

jveitchmichaelis, 2023-08-04:
Assuming you've created a camera object, give this a go. This should change the video mode to TLinear.

```
function_id = 0x000E000E #sysctrlSetUsbVideoIR16Mode
command = struct.pack(">H", 2) # FLR_SYSCTRL_USBIR16_MODE_TLINEAR = 2
res = camera._send_packet(function_id, data=command)
res = camera._decode_packet(res)
```

This should put your camera into TLinear mode and then multiply your raw values by 0.01 to get to Kelvin.

jveitchmichaelis, 2023-08-23:
@bakas90 I see the thumbs up, any luck with this? :)

olfk1234, 2024-03-27:
@jveitchmichaelis Hello! I tried the code above and I got a warning stating "WARNING:flirpy.camera.boson:Boson response: bad command ID" when running the decode packet function. Is there another workaround to convert the Boson into TLinear mode or to get the temperatures using flirpy?

jveitchmichaelis, 2024-03-28:
If you go to the receive function and print/dump the return from the camera, what do you get?

I don't have a radiometric Boson so I'm unable to check this myself.

olfk1234, 2024-03-28:
@jveitchmichaelis After doing `print(camera.receive())` after the code that you mentioned before, I am getting `b' '`

jveitchmichaelis, 2024-03-28:
Try with `0x000E000D`:

[image]

Looks like a typo with the previous function code (that was get, not set).

millercommamatt, 2025-03-20:
I'm just diving into flirpy for my radiometric Boson. I requested the USB Vidieo IR Mode as show below. The returned status value was 2 which should be T-Linear mode. Grabbing an image resulting the an unsigned 8-bit numpy array which doesn't seem correct to me. I was expecting a uint16 array.

I'm sure I'm missing something simple, but this is my first time both with this library and more importantly, a Boson camera.

```
>>> res=camera._send_packet(0x000E000E)
>>> res=camera._decode_packet(res,receive_size=4)
>>> res
b'\x00\x00\x00\x02'
>>> image=camera.grab()
>>> image
array([[ 93, 117,  93, ..., 117,  86, 117]],
      shape=(1, 655360), dtype=uint8)

>>> res=camera._send_packet(0x003E0002)  # TLinearGetControl()
>>> res=camera._decode_packet(res,receive_size=4)
>>> res
b'\x00\x00\x00\x01'  # 1 = FLR_ENABLE

```

jveitchmichaelis, 2025-03-20:
Yeah though you shouldn't need to use the low level functions to do this. Interesting that grab is not returning a 16 bit array though, it should handle that conversion for you...

millercommamatt, 2025-03-20:
I'll dig into why OpenCV isn't returning the expected data structure. Issue #105 reported the same phenomenon. Let me know if you have troubleshooting suggestions. I've not tangled with OpenCV before.

I'm using a Radiometric Boson 640.

Thanks for this library. It saved me a ton a work. I might send some PRs to expand some of the functionality for the Boson class.

jveitchmichaelis, 2025-03-20:
Yeah it looks like a regression in OpenCV so you could try and bisect to figure out where it was introduced and we could raise an issue directly.

Otherwise I would really welcome PRs for radiometric functionality as this is something I don't have the ability to test at the moment.

Thanks!

jveitchmichaelis, 2025-03-22:
By the way you can also try casting the array directly using `view`: https://numpy.org/doc/2.2/reference/generated/numpy.ndarray.view.html

millercommamatt, 2025-03-24:
Perhaps this is the origin of the apparent change in OpenCV behavior?
https://github.com/opencv/opencv/pull/18498/files

Note the changes in output data types for `case V4L2_PIX_FMT_Y16:`...

jveitchmichaelis, 2025-03-24:
Interesting. I'll have a look. So are you able to confirm whether a build either side of that commit passes/fails?

What's unclear to me without digging is why we're getting a flat array back and not just some messed up values (like if the endianness is backwards). Grab should return a 2D image if the VideoCapture object has some idea of width/height. What happens if you query the capture properties with a build of OpenCV that gives you the wrong shape output?

jveitchmichaelis, 2025-04-03:
@millercommamatt could you confirm this fix please?

https://github.com/opencv/opencv/pull/27193