LJMUAstroecology/flirpy #91: Duo R Support

mcherep, 2023-09-21:
I'm sorry if I'm missing something obvious. The documentation says that the library has been tested on FLIR Duo Pro R, but I can't find any examples. I only need to grab an image and return the average temperature.

jveitchmichaelis, 2023-09-21:
The Duo doesn't support live streaming over USB, but flirpy can process the saved SEQ files. This is a limit of the camera unfortunately.

mcherep, 2023-09-21:
Thanks for replying so quickly, it's great (and sad) to know.

closed 2023-09-21 by mcherep

jveitchmichaelis, 2023-09-21:
If you just need to grab an image, then either the Lepton or Boson-R would do, depending on whether you need the resolution (or if you also need an RGB image, but that can be a lot cheaper to add on).

mcherep, 2023-09-21:
Thank you so much! I have some cameras around, so I'll try to go with these.