LJMUAstroecology/flirpy #93: All data read from SEQ file is zeroed

DrBwts, 2024-05-07:
I'm trying to use `split_seqs.py`

The meta data on the camera is right & the number of frames I was expecting is correct. During acquisition I could see the thermal mapping but when the data is split all *.jpg and all *.tiff files just contain 0 at every pixel position in every rendered frame.
  
Calling the script using,

`pyton split_seqs.py -i Rec_Fives_3lve_mk1_spindletest1_120_10_07_20_473.seq --split_filetypes`

jveitchmichaelis, 2024-06-22:
Hi, sorry I missed this - are you using the version in the repository (not from pip)? If you're able to share a SEQ file that would help debug. It's odd that the extraction works at all if the frames are empty.

jveitchmichaelis, 2024-07-03:
Feel free to open if you need more support but I'm tidying up older issues.

closed 2024-07-03 by jveitchmichaelis