LJMUAstroecology/flirpy #94: Unable to extract per-pixel raw temperature

juuiceboxx, 2024-06-21:
I am unable to collect the metadata (mostly need timestamp) of each frame of my seq file, and I really need to be able to collect a matrix or csv that contains the temperature values associated with each pixel of every frame of the video. I have attempted to mess with the "export_raw" argument in split_seqs, but have not been able to get the txt file to be exported. I think there may be an issue with the txt file being overwritten by the .fff files, because they have the same folder name ("raw"). 

I have also attempted to extract the temperature data from the radiometric tiff files, but have found that the tiff's only contain what I believe is the 16-bit normalized temperature values from 0-~65,000. I also noticed the raw2temp function is used to calculate the explicit temperatures, so I am wondering if there is a way to convert the normalized values from the radiometric tiffs to the actual values in celsius or kelvin again using that function?

jveitchmichaelis, 2024-06-22:
The radiometric tiffs have scaled absolute values for space efficiency and
portability - ie we don't store as float. The metadata does not contain per
pixel values, it just contains the calibration factors for your camera (so
really you only need one copy).

We don't currently offer a CSV export. Can you explain how you'd use it?
This is a fairly common request but I'm reluctant to implement it because
it's a very inefficient storage medium for images.

The conversion factor is 0.04 K per count. So read the image as a 16-bit
array and multiply by that.

Let me know if you have any other questions!

jveitchmichaelis, 2024-06-22:
Timestamp though, if you need that for every frame you could load the FFF
files into flirpy (just pass the filename to the constructor) and query the
`meta` dictionary.