LJMUAstroecology/flirpy #96: FLIR ONE Edge Pro

mikelmujika11, 2024-07-03:
Hi,

I am interested on buying a thermal camera but before buying it I need to ensure that I am able to acces the camera info (frame + temperature values) via Python using OpenCV. Does this library support this type of camera?

Thanks

jveitchmichaelis, 2024-07-03:
Flirpy does support this but not for the Edge Pro. You would need to use one of the "camera cores" like the Lepton or Radiometric Boson.

mikelmujika11, 2024-07-03:
I thought they were like the TOPDON cameras that in addition to connecting to the cell phone can also be connected to the PC. Do you know of any Flir cameras that are currently sold, with similar features to the Edge Pro, that can be connected to the PC and accessed?

jveitchmichaelis, 2024-07-03:
Sorry, no idea about TOPDON. Either way, at this time Flirpy does not support access. It's possible that the camera is a UVC device, in which case you could retrieve imagery but I doubt it returns thermometry without a library.

What features do you need? Happy to give suggestions but I'd need specific requirements. As I suggested, both the Lepton and Boson R return radiometric images and both work with OpenCV through various means. Most of these smaller consumer cameras use a sensor like the Lepton - eg 320px, 9fps. You don't get RGB/MSX but that's cheap to add with a second camera.

Your other option is to look at FLIRs APIs (see Atlas) if they exist and write your own wrapper to talk to OpenCV.

https://www.flir.com/developer/mobile-sdk/ it exists for mobile at least.

Or just buy a TOPDON and use a different library - I'd be happy to work to add support but as I don't own one, I can't do any physical testing.

See https://github.com/leswright1977/PyThermalCamera

mikelmujika11, 2024-07-03:
I have found FLIR to be quite limited so I will look for another brand.

thanks for everything

jveitchmichaelis, 2024-07-03:
No problem, I don't have any affiliation with any of these companies. If you do buy something, happy to try and add support into Flirpy if we can get it going.

I would say that FLIR's camera cores tend to be pretty good, if expensive, and the interfaces are well documented.

closed 2024-07-03 by jveitchmichaelis