LJMUAstroecology/flirpy #97: How to convert thermal image to temperature values?

aatefi2, 2024-07-08:
Hi,

I have a MicaSense Altum-PT camera which uses "320 x 256 FLIR Boson" as its thermal sensor. The saved thermal images are completely gray color. The raw temperature data is 16 bit (for example: 31399, 31436, 31415,...). From the metadata of the thermal image, I was able to extract 6 values for `Thermal Calibration Parameters  : 1, 315.60000000000002, 1192951.5, 1509.95947265625, 1, 13831.46484375.` using `meta.get_item('XMP:ThermalCalibrationParameters'`.
I am wondering is there any python code from `flirpy` to calculate the temperature of all pixels of an image? How can I interpret these 6 values of Thermal Calibration Parameters for that code? are the calculated temperature values compensated for atmospheric/air temperature?

Thank you,
Abbas

jveitchmichaelis, 2024-07-09:
MicaSense has some info here:
https://support.micasense.com/hc/en-us/articles/360036377834-Altum-and-Altum-PT-Thermal-Band-FAQ
see
if that helps!

Most thermal cameras output Kelvin with a scale factor. You shouldn't need
to worry about the onboard calibration unless you have concerns that what
you're imaging has a significantly different emissivity or as you say. You
can look into the Boson-R software IDD document which should have more
detail on the metadata.

https://f.hubspotusercontent10.net/hubfs/20335613/boson-sdk-documentation-february-2019.pdf

> When processing your thermal data in Pix4D and Agisoft, you may notice
that your thermal (LWIR) band output has much higher than expected values.
This is because the output units are centi-Kelvin. In order to convert the
temperature values from centi-Kelvin to a unit that is easier to interpret,
you will need to divide the output by 100 (which gives you Kelvin) and then
subtract 273.15 to convert to Celsius.

aatefi2, 2024-07-09:
> MicaSense has some info here: https://support.micasense.com/hc/en-us/articles/360036377834-Altum-and-Altum-PT-Thermal-Band-FAQ see if that helps! Most thermal cameras output Kelvin with a scale [...]

Hi jveitchmichaelis,

Thank you for your comment. I looked at the links, but still could not find a detailed information to map thermal image to pixel temperature values using a python code. I just found this link https://flir.custhelp.com/app/answers/detail/a_id/3321/%7E/flir-cameras---temperature-measurement-formula, but still I am not sure how to use these extracted values in the formula from that link. The issue is, I have already asked MicaSense in this regard, but they suggested me to get the calibration equation from FLIR; I asked FLIR, but they did not answer me yet and I could not find any information for FLIR BOSON camera in this matter.

jveitchmichaelis, 2024-07-09:
Unless I'm missing something obvious, have you tried making that conversion?

```python
temps = (frame.astype(np.float32)/100.0) - 273.15
```

The images should already be calibrated.

aatefi2, 2024-07-09:
> Unless I'm missing something obvious, have you tried making that conversion?
> 
> ```python
> # Assume frame is a 16-bit numpy array with your data
> 
> temps = (frame.astype(np.float32)/100.0) - 273.15
> ```
> 
> The images should already be calibrated.
> 
> `(31399/100)-273.15` would map to about 40 C if that's a sensible value for what you're imaging.

Thank you for your prompt reply. I think your equation is correct since it gives me similar results to temperature data from another camera which had overlap with MicaSense images. The values from that camera is high as well (for example 38, 39, 40,41,... C).

jveitchmichaelis, 2024-07-09:
Great, this is just using what MicaSense says in their documentation. The Boson itself has a huge array of radiometry settings that currently aren't supported by Flirpy (something like 60 pages of documented commands), but in most cases you don't need to touch it because the camera will output (scaled) temperature values when in `TLINEAR` mode.

closed 2024-07-18 by jveitchmichaelis