LJMUAstroecology/flirpy #99: problems installing flirpy 4.0

ayazlali, 2024-07-16:
Hi! 
I have Anaconda and Spyder to run Python 3.11 on Windows 10 enterprise.
I tried installing flirpy using pip (```pip install flirpy```); the installed version was 3.0 and I ran into some problems when anaylzing .seq files. I got an "assertion error" and found in a different thread that this is now solved in version 4.0. I couldn't uninstall flirpy with pip (tried ```pip uninstall flirpy```, cancelled after nothing happened after 1 hr ). Next, I created a new environment to try using the ```python setup.py install``` method mentioned in the installation part of the documentation. I downloaded the latest release (4.0), however I couldn't find "setup.py" in the folders. I am very new to python building environments and there must be something I am missing, could you please help me with installing flirpy 4.0?

jveitchmichaelis, 2024-07-17:
If you've downloaded the files, try running `pip install .` from the repository folder.

ayazlali, 2024-07-17:
I did that and it worked! I also ran ```pip install -r requirements.txt``` to install all the dependencies and was able to decrypt a .seq video. Thank you so much for Josh @jveitchmichaelis for the guidance and for making this repository!

jveitchmichaelis, 2024-07-18:
Great, thanks!

closed 2024-07-18 by jveitchmichaelis