LJMUAstroecology/flirpy #101: split_seqs struct.error: unpack_from requires a buffer of at least 32 bytes for unpacking 32 bytes at offset 0 (actual buffer size is 0)

ww895543212, 2024-07-25:
[image]

jveitchmichaelis, 2024-07-27:
Please try installing from the repository directly and see if this still occurs.

nysan, 2024-09-24:
I see this on 0.4.0 as well, in an python3.12 venv

jveitchmichaelis, 2024-09-24:
Thanks for the additional report. If you're able to share an example input that fails it would be very useful. Otherwise I have no way to test/reproduce to figure out where the bug is (similarly is this a regression, did it work in previous versions?)

I'm extremely busy at the moment but I'm hoping to have a look at this in the next few weeks (or feel free to debug and PR!)

Thanks again

ww895543212, 2024-09-26:
I have uploaded the problematic file to Google Drive.
[https://drive.google.com/file/d/1VqCteOw05LSW08laQaV0zCn06U5sBYZ3/view?usp=drive_link](url)