LJMUAstroecology/flirpy #102: Image distortions - Dark horizontal lines

MDeBSAL, 2024-08-13:
I am using a Raspberry Pi 5 with the latest FlirPy version. I am getting short dark lines in the images which only appear when using the Raspberry Pi. On Windows systems the error does not occur. Any thoughts what could be the issue

[image]

jveitchmichaelis, 2024-08-13:
This is probably an issue relating to OpenCV. Flirpy wraps OpenCV's video functions to read images and does not implement any drivers itself, so you should be able to reproduce this by writing a simple script to capture a frame from the camera.

Have you tried visualising the video stream with other software? That should also be possible.

I you believe this is a bug, please update your post using the [issue template](https://github.com/LJMUAstroecology/flirpy/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=)

MDeBSAL, 2024-08-13:
As mentioned, when using a Windows system the code works perfectly fine. It just happens when working on a Raspberry Pi.

jveitchmichaelis, 2024-08-13:
Windows uses a different capture (typically DirectShow) backend to Linux (e.g. V4L). You can confirm where this issue is introduced by capturing a frame from OpenCV without using Flirpy.

MDeBSAL, 2024-08-13:
Thank you for your valuable input. It seems that it is an issue with OpenCV. I have used this script and got the the same result.

import cv2

cap = cv2.VideoCapture(0)

if not cap.isOpened():
    print("Error: Could not open camera.")
    exit()

while True:

    ret, frame = cap.read()

    if not ret:
        print("Error: Can't receive frame (stream end?). Exiting ...")
        break

    cv2.imshow('Live Video Feed', frame)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
cap.release()
cv2.destroyAllWindows()

[image]

jveitchmichaelis, 2024-09-02:
Great, closing as this is an issue with opencv on RPi.

closed 2024-09-02 by jveitchmichaelis